UtilitechAS/amsreader-firmware #261: Tid på data i json er offset 2t på realtime calculation today

monsivar, 2022-04-04:
Når jeg ser på data fra http://<ams_hostname>/data.json, skifter ikke døgnet før midnatt+2 for følgende

"d" : {          # Real time calculation today
            "u" : %.2f,  # Consumption
            "c" : %.2f   # Cost
        },

Kjører 2.1.1

gskjold, 2022-04-04:
Tidssone satt opp i Config-> NTP? Er klokken på dashbordet korrekt?

monsivar, 2022-04-04:
Klokka er riktig ja :-), og det var den også når jeg hadde kun "Obtain NTP server from DHCP" huket av, så er jeg avhengig av å sette UTC+1 og sommertid?

Mons

gskjold, 2022-04-04:
Det er nødvendig ja. Den får kun tid fra NTP, tidssonen er ikke oppgitt, selv ikke med DHCP. Skal sette opp en test hjemme og se om jeg får samme feil

monsivar, 2022-04-05:
Justerte tidene og en restart i går, ved midnatt rapporterte den riktig. så da ble jeg lurt av at gui viste riktig tid selv om verken tidssone og sommertid var satt