UtilitechAS/amsreader-firmware #262: Priser via MQTT (Home Assistant auto discovery)

Handl3vogn, 2022-04-04:
**Describe your problem**
Ser ikke ut som jeg får noen priser over mqtt (Home Assistant auto discovery).
Har satt opp ENTSO-E API og får priser i web grensesnittet men får dem ikke over til home-assistant.

**Hardware information:**
 - Meter: [e.g. Kamstrup]
 - AMS reader: [e.g. Pow-K, ESP32]
[image]
[image]

gskjold, 2022-04-05:
For anyone else visiting this issue, I can confirm that HA autodiscovery for prices have not been implemented. As I do not have a HA setup and do not have time to dive in to this, feel free to provide a working solution. If you do not have coding experience, auto discovery JSON for each necessary sensor is sufficient.

PalmMx, 2022-04-05:
Energy prices in Home Assistant can be retrieved using the Nordpool component:

https://github.com/custom-components/nordpool

DarkFox, 2022-06-27:
As I understand it, Nordpool have deprecated their API, and there's a good chance it'll be discontinued or break entirely, soon.

On-topic: Is the price still sent when in Home Assistant mode (but just not auto discovered), or would I have to switch to JSON mode?

DarkFox, 2022-06-27:
Looks like I was on an older version, and that's why the prices weren't showing up.

However, the Home Assistant feed, seems to be missing a number of things.

- It only includes the next 12 hours of prices (out of 22 hrs total, at the time I tested).
- No real time price calculations.
- The messages don't seem to be retained (Not sure if that's just the prices, or all of them).

```
Besked 327 er modtaget på kamstrup/prices kl. 01.38:

{
    "id": "58:CF:79:9F:B6:24",
    "prices": {
        "0": 1.962,
        "1": 1.9201,
        "2": 1.9057,
        "3": 1.8939,
        "4": 2.0005,
        "5": 2.462,
        "6": 2.6632,
        "7": 2.7608,
        "8": 2.592,
        "9": 2.4742,
        "10": 2.1837,
        "11": 1.994,
        "min": 1.8668,
        "max": 3.2997,
        "cheapest1hr": "2022-06-28T11:00:00Z",
        "cheapest3hr": "2022-06-28T11:00:00Z",
        "cheapest6hr": "2022-06-28T09:00:00Z"
    }
}
```

gskjold, 2022-06-28:
Will look into getting more information in HA mode. The price JSON was limited to 12 hrs because of memory constraints with ESP8266, but it could be that it can be expanded now with the larger buffer. Raw mode has all available hours in separate topics.

DarkFox, 2022-06-28:
I'm slightly confused, why you wouldn't just use an array for the values? That would save quite a lot of characters and make it easier to use in most applications.
Like so:
`"future_prices": [1.962, 1.9201, 1.9057, 1.8939, 2.0005, 2.462, 2.6632, 2.7608, ....],`

gskjold, 2022-09-28:
Work in progress. Currently implemented auto configuration for sensors containing price and realtime data.

What I need to verify:
- The new firmware functions with device configured from old auto config data (Version 2.1.7 or below)
- The new price sensors are functioning as expected (Has future price 0-11, min/max price ahead and 1hr, 3hr and 6hr cheapest times ahead)
- Real time sensors functioning as expected (Has month max, threshold, peaks, hour/day/month use/cost/produced)

WARNING: Use the DEV firmware below at your own risk. You need to be able to flash by USB/TTL if firmware does not work.
[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/9667912/esp32.zip)
[esp32s2.zip](https://github.com/gskjold/AmsToMqttBridge/files/9667913/esp32s2.zip)
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/9667914/esp8266.zip)

gskjold, 2022-09-30:
Installed HA on VirtualBox for testing, made some adjustments:
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/9686486/esp8266.zip)
[esp32s2.zip](https://github.com/gskjold/AmsToMqttBridge/files/9686487/esp32s2.zip)
[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/9686488/esp32.zip)

closed 2022-09-30 by gskjold

klem-everywhere, 2022-10-03:
Looks great! I'll test it later this week.
Are you planning of exposing all 22 hrs prices? Either as a list or only enable for esp32?

gskjold, 2022-10-06:
Will expose all hours in a future release