UtilitechAS/amsreader-firmware #263: Support for update entities in Home Assistant introduced in 2022.4

bjornsivertsen, 2022-04-06:
As a person that rarely visits the AMS interface, but instead pulls the data into my Home Assistant instance through MQTT, I might miss important updates.
From version 2022.4, [Home Assistant will support update entities](https://rc.home-assistant.io/blog/#introducing-update-entities).

This could allow the visibility within Home Assistant, if an update is available.

In addition, and IF(!) AMS will support 1-click-to-update (rather than uploading a downloaded firmware and installing), you would be able to initiate the update from Home Assistant.

gskjold, 2022-04-06:
ESP32 version of the firmware can self update. If anyone can provide the necessary JSON to publish on MQTT, I can implement the changes.

bjornsivertsen, 2022-04-06:
> ESP32 version of the firmware can self update. If anyone can provide the necessary JSON to publish on MQTT, I can implement the changes.

Is the same functionality planned for ESP8266? Although I am more than happy with the current approach.

gskjold, 2022-04-06:
Yes, but I've had some issues getting it to work properly. I have not given up though.

StarkJohan, 2022-08-31:
Please make the auto updating optional and not forced when implemented, it can surely be the default but if running snapshots or modified versions forced updates is undesirable. Same goes for updates that might break some configuration.

gskjold, 2022-09-30:
I cannot find any specific examples on how this should be published on MQTT. If someone can reference a proper example, I will look into this.

bardahlm, 2022-11-03:
There are examples here: https://www.home-assistant.io/integrations/update.mqtt/

gskjold, 2025-02-02:
Has anyone made this work for something else before? I am trying to send this entity configuration to MQTT:
`{"name":"Firmware","stat_t":"amstest/firmware","uniq_id":"ams111_FW","obj_id":"ams111_FW","dev_cla":"firmware","cmd_t":"amstest/command","pl_inst":"fwupgrade"}`

But it throws an error:
`Error 'expected SensorDeviceClass or one of 'date', 'enum', 'timestamp', 'apparent_power', 'aqi', 'area', 'atmospheric_pressure', 'battery', 'blood_glucose_concentration', 'carbon_monoxide', 'carbon_dioxide', 'conductivity', 'current', 'data_rate', 'data_size', 'distance', 'duration', 'energy', 'energy_storage', 'frequency', 'gas', 'humidity', 'illuminance', 'irradiance', 'moisture', 'monetary', 'nitrogen_dioxide', 'nitrogen_monoxide', 'nitrous_oxide', 'ozone', 'ph', 'pm1', 'pm10', 'pm25', 'power_factor', 'power', 'precipitation', 'precipitation_intensity', 'pressure', 'reactive_power', 'signal_strength', 'sound_pressure', 'speed', 'sulphur_dioxide', 'temperature', 'volatile_organic_compounds', 'volatile_organic_compounds_parts', 'voltage', 'volume', 'volume_storage', 'volume_flow_rate', 'water', 'weight', 'wind_speed' for dictionary value @ data['device_class']' when processing MQTT discovery message topic: 'homeassistant/sensor/ams111_FW/config', message: '{'device_class': 'firmware', 'object_id': 'ams111_FW', 'command_topic': 'amstest/command', 'name': 'Firmware', 'payload_install': 'fwupgrade', 'unique_id': 'ams111_FW', 'state_topic': 'amstest/firmware'}'
`

gskjold, 2025-02-03:
Nvm, I'm just an idiot 😆 

[image]

closed 2025-02-03 by gskjold

bjornsivertsen, 2025-02-03:
Fantastic you got it working, looking forward for the update dropping!