UtilitechAS/amsreader-firmware #264: Support for further Austrian (and Luxembourg) meter

grubmueller, 2022-04-06:
Hi!

First of all congratulation to the really nice project.
I'd like too ask if it would be possible to provide support for a further Austrian Smart Meter.
(Yes, Austria is a small country, but we have more than 10 net suppliers and they really managed to implement different smart meter solutions!)
I'm in Graz, Styria (South-East of Austria), and I've a Sagemcom smart meter with a DSMR-P1 hardware interface (no MBUS).

*** What I got so far:
- I managed to get an encryption and an authentication key from my net provider (the hardest step so far)
- I could read and decrypt the data with a simple UART to USB receiver (115200 baud)
- I flashed a D1 mini with your firmware and could connect to it, but no reading meter reading could be done
- The telnet debugger tells something about an unknown starting character "DB" with an interval of 10s (push out rate of my meter)
- Even though, I think this confirms that my hardware connection/communication works, because "DB" really is the start signal of the used protocol.
- I attach a message from my meter. As you can see it is very similar to those of Sweden you have posted in your repository, but without all this MBUS info at the beginning. My meter start right away wit the "DB"
- After some further web research I found out that this protocol should be the same as used in Luxembourg, see [Link](https://www.weigu.lu/tutorials/sensors2bus/04_encryption/index.html). So maybe a solution would be interesting not only for this very small Austrian region.

Do you see any chance to extend your software to support this meter?

Regards, Michael

[smart_meter_data_encrypted comment.txt](https://github.com/gskjold/AmsToMqttBridge/files/8429053/smart_meter_data_encrypted.comment.txt)
[smart_meter_data_decrypted.txt](https://github.com/gskjold/AmsToMqttBridge/files/8429054/smart_meter_data_decrypted.txt)

ArnieO, 2022-04-06:
I think you need to invert the data signal (see config/meter).

grubmueller, 2022-04-06:
I'm aware of that, signal is already inverted in hardware, "invert pin" feature in software is left unchecked (with 115200 baud and 8N1 parity).
I am quite sure that the hardware configuration should work, because
- I use a 3V3 serial to USB converter in parallel for debugging which delivers valid data
- I flashed an ESPhome firmware to the same setup and got valid data (up to a certain point, further I had the problem that the used information parser library couldn't deal with the "Wh" units from my meter instead of the standard "kWh" units)
- The exact debug output is "(W) (readHanPort) Unknown data format DB"; 0xDB really is my start byte; for me this also looks like the device got the right data but doesn't know how to deal with it?
Any further hints?

gskjold, 2022-04-07:
Cool! This could possibly be the first DSMR "in the wild" to use this software, at least with encryption. I have not yet implemented decryption for this standard. The reason being that I have not had anything to test with. If you feel comfortable sharing a couple of frames and your encryption pair, email me at gunnar.skjold@gmail.com . If not I can try to implement without and have you test it. I see it is the same used with MBUS standard, so it should be pretty straight forward.

grubmueller, 2022-04-07:
Thanks so far. Just sent you a mail with a frame and my encryption pair. Will provide some more frames on weekend.

closed 2022-04-23 by gskjold