UtilitechAS/amsreader-firmware #267: SAGEMCOM T211

christianritella, 2022-04-20:
From a closed issue, #253 
https://github.com/gskjold/AmsToMqttBridge/issues/253

I have connected RX to D7 and HAN=UART2/GPIO13.
Baudrate 115200 Parity 8N1
I have tried invert on/off;
Invert off: (W) (readHanPort) Unknown data format 40
invert on: (W) (readHanPort) Unknown data format 3F
Does this indicate that it is receiving data?
Is it possible to monitor received data from debug?

I have no resistors connected.

Any other suggestions?

ArnieO, 2022-04-21:
Any other suggestions?

Please try a Telnet debug:
System/Debugging: Select Telnet and Verbose mode
Open a command window on your PC, and write "telnet 'IP-address of device' "
You may need to activate telnet, as it is by default disabled on Windows (how to do that: Google).
Then paste here what you get.

gskjold, 2022-04-21:
You are definitely receiving data, and like @ArnieO suggested, it would be useful if you run verbose logging so that we can see a full datagram to try determine if we are receiving an unknown payload or if there is some kind of data corruption.

christianritella, 2022-04-23:
Here is the output from Verbose;

(D) (readHanPort) Frame dump (1b):
(D) 3F
(W) (readHanPort) Unknown data format 3F
(D) (loop) Used 0 ms to update temperature
(D) (readHanPort) Frame dump (1b):
(D) 3F
(W) (readHanPort) Unknown data format 3F
(D) (loop) Used 0 ms to update temperature
(D) (readHanPort) Frame dump (1b):
(D) 3F
(W) (readHanPort) Unknown data format 3F
(D) (loop) Used 0 ms to update temperature

ArnieO, 2022-04-25:
So did you try to invert the signal?
[image]

ArnieO, 2022-04-25:
Could you send screenshot of your Config/meter and System/GPIO screens?

christianritella, 2022-04-25:
Debug with was with invert, same result with it off.
[image]
[image]
[image]

ArnieO, 2022-04-25:
Thank you; I see no issues here. 

I'm re-reading your previous messages now and see that I don't quite understand this: "I have connected RX to D7 and HAN=UART2/GPIO13." So maybe there is some misunderstanding here?

You should connect the DATA pin from the P1 connector to UART2/GPIO13, which should be D7 pin on your board. In order to receive data from a P1 port you also need to pull D_Rqust high, please see below snippet of a circuit diagram for a device connecting to P1 port.

**Important note:** The DATA output on the P1 port is Open Collector, meaning you need a pullup resistor on in to 3,3V as indicated (R5 in the below diagram).

[image]

christianritella, 2022-04-25:
I think I understand, give me a couple of days...

ArnieO, 2022-04-27:
Great! 👍 
The red ESP is an indication your voltage is low, or to be exact: that the voltage as read by the firmware is low. 
[image]

In System / GPIO you need to enter the resistor values of the voltage divider on your NodeMCU into "GND resistor" and "Vcc resistor", and select the correct GPIO input in "Vcc".
These fields:
[image]

This page indicates that the values are 100k and 220k but please check if that is correct for your board. 
https://circuitsgeek.com/guides-and-how-to/esp8266-adc-tutorial-read-nodemcu-analog-input/

ArnieO, 2022-04-27:
With correct values you will then get close to correct value. You can calibrate it by adjusting "Multiplier" on the same page. 

Measure Vcc on your board, and adjust multiplier until you read correct voltage.

gskjold, 2022-04-28:
> This page indicates that the values are 100k and 220k but please check if that is correct for your board

Although this is correct, the voltage divider on NodeMCU is intended for ADC input, which is disabled in this firmware since ADC_VCC is enabled. When ADC_VCC is set, nothing should be connected to ADC pin, but as you have pointed out there is on NodeMCU. This interferes with internal Vcc sensor, so the only way to get a semi-accurate reading is to adjust Vcc multiplier until you get a sane value. Guessing 1.11 (which is what I use for D1 mini)

ArnieO, 2022-04-28:
> Although this is correct, the voltage divider on NodeMCU is intended for ADC input, which is disabled in this firmware since ADC_VCC is enabled. When ADC_VCC is set, nothing should be connected to [...]

OK, thank you for clearing up my misconception! 
I believe the reason this inconvenience cannot be managed in code is that ADC read from GPIO pin vs ADC reading from internal Vcc is a selection that must be done at compilation/build level - right?

gskjold, 2022-04-28:
That is correct, we would need to have a "ADC" version of the ESP8266 firmware to resolve that issue. Not too sure it is that important. The multiplier adjustment is a usable fix I think.

ArnieO, 2022-04-28:
> That is correct, we would need to have a "ADC" version of the ESP8266 firmware to resolve that issue. Not too sure it is that important. The multiplier adjustment is a usable fix I think.

Definitively not worth a separate version, your workaround is the way to go.

closed 2022-06-04 by gskjold