UtilitechAS/amsreader-firmware #268: mqtt ssl

1968-Larry, 2022-04-27:
Adding PEM certificates/keys results in 404 error on page /config-mqtt
Hardware esp8266 release 2.1.2
If you go back and upload all certificates/keys it looks like files has been uploaded. When you save the settings it starts connecting and the esp reboots constantly. 

Running esp32 release 2.1.2
If you go back and upload all certificates/keys it looks like files has been uploaded.

Serial output: MQTT_connect)(C1) Failed to connect to MQTT: -3

MQTT server: 
OpenSSL Error[0]: error:14094412:SSL routines:ssl3_read_bytes:sslv3 alert bad certificate

Config download doesn't contain any certificates/keys, not sure if it should.

gskjold, 2022-04-28:
> Config download doesn't contain any certificates/keys, not sure if it should

It is not included in config download, maybe in the future.

Is the clock on the module correct?
Does the hostname you entered in "Host" match the one used in the certificate?

1968-Larry, 2022-04-28:
Clock is correct
hostname doesn't match, certificate contains domainname and not possible to add in configuration as hostname. But should upload even with correct name result in 404? CA i guess is the mqtt servers root ca and it also responds 404 when uploaded. I think privatekey shouldn't be allowed to download but hard to know if files was stored correct. Perhaps add some information like subject common name should be displayed

gskjold, 2022-04-28:
I agree it should not result in 404, I will look into that. Hostname has to match for it to work as it is implemented right now. I could possibly include a "insecure" checkbox that will ignore the name. I will look into extracting information from the certificate for debugging purposes. Not sure how easy that is, but a really good idea. As for including private key in config download, I think it could be included when "Include secrets" is checked.

1968-Larry, 2022-04-28:
I found the problem, I thougt you ment that AMS validated the uploaded certificate against it's own hostname. Was to focused on the upload problem. Changing hostname to mqtt server and the communication works fine.

ESP32 works fine except 404 message, esp8266 needs to upload new firmware. Can't tell why it's rebooting constantly.

gskjold, 2022-04-28:
Alright, good to hear, I will set up a SSL test at home one day to find out why ESP8266 does not work with SSL. But it could be that it does not have enough memory to perform SSL for MQTT, the memory consumption of the firmware is already way past the limit for which HTTPS needs to work.

The 404 issue should be easy enough to fix :)

gskjold, 2022-06-04:
Fixed 404 issue for next patch. I am unable to use MQTT SSL with ESP8266, I'm assuming we have just run out of memory to do it, so I have disabled support in next patch.

closed 2022-06-04 by gskjold