UtilitechAS/amsreader-firmware #273: Possibility to add DLMS / KMp protocol

chmadkk, 2022-05-03:
Hi. 
i was sondering if theres a posibillity to add the KMp protocol which uses request communication instead of push ?

ArnieO, 2022-05-03:
Is that the Kamstrup protocol you're referring to, where the device must send a command to a Kamstrup power meter to request data? If so - the answer is No; unless you have access to the protocol. To my knowledge it is a Kamstrup proprietary protocol, not publicly available. Which is why the better answer to your situation would be to request your grid company to upgrade the meter (can be done remotely) to push data. 

Please consult [this thread](https://github.com/gskjold/AmsToMqttBridge/issues/126) for more information.

gskjold, 2022-05-05:
KMP is a proprietary protocol and will not be implemented I'm afraid

closed 2022-05-05 by gskjold