UtilitechAS/amsreader-firmware #275: Old Kamstrup 382M Power meter

tjay78, 2022-05-06:
Hi
it is a super cool project and I would like to use it, but my problem is that I have an old model Kamstrup 382M Power meter that many here in Denmark have installed in their homes..
I know it is without encryption key and you have to ask for data, it does not send every 10sec. itself, 
and there are no projects that have solved this yet because it is a close protocol until I came across this project, 
There is one who has made it work by using a Raspberry PI and a USB to RS232 converter and some python code...
So my request is it possible to use some of his code to make it work in AmsToMqttBridge??
That could be awesome :)

Link : https://github.com/MichaelRosenlund/382M_to_mqtt

tjay78

ArnieO, 2022-05-06:
Hi, 

Thank you for your kind words, but we are not planning to implement in our code the Kamstrup proprietary protocol necessary for reading those old meters. Our understanding is that those meters are in the process of being replaced in Denmark, you might want to check with your grid operator (nettselskab): https://www.danskenergi.dk/vejledning/nettilslutning/find-netselskab

As far as I know, the only available solution for those meters is the Smart-Me device, which is a cloud- and app-based solution. https://www.elmodul.dk/product-page/smart-me-kamstrup-modul

tjay78, 2022-05-07:
hi,

I'm sorry to hear that..
Because the replacement of the Power meter will not be until 2030 according to my grid operator... :(
And about Smart-Me device who buys this ??? I mean cloud solution with who?? and the price is the same as a new Power meter costs....

tjay78

gskjold, 2022-05-11:
This is a undocumented proprietary protocol. The fact that someone have reverse engineered this to such a degree is impressive, but not useful enough to implement efficiently without having such meter at hand myself. I also question the legality of such implementation. I'm not saying I think it is illegal, but when you implement code from reverse engineered knowledge, there is always the chance you violate a patent. 

In short, get Kamstrup to release me the documents and give the green light for having it implemented in an open source project and I'll do it.

closed 2022-06-04 by gskjold

ArnieO, 2024-06-04:
@tjay78 : https://github.com/UtilitechAS/amsreader-firmware/discussions/725#discussioncomment-9663204