UtilitechAS/amsreader-firmware #278: Flashing ESP32. Which method?

mrwee, 2022-05-10:
The [WiKi](https://github.com/gskjold/AmsToMqttBridge/wiki/flashing) states

```
esptool.py--chip esp32 --port <port> --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 bootloader.bin 0x8000 partitions.bin 0x10000 firmware.bin
```
so:
0x1000 bootloader.bin
0x8000 partitions.bin
0x10000 firmware.bin
for ESP32.

However, the "flash.sh" file, included with all the binaries, states
```
$esptool --chip esp32 --port $2 --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect \
    0x1000 bootloader_dio_40m.bin \
    0x8000 partitions.bin \
    0xe000 boot_app0.bin \
    0x10000 firmware.bin
```

so there's also a **"boot_app0.bin"** included in this example.

Since I'm struggling to get Wi-Fi to work on two ESP32 modules, I just wondered which method is corrrect? (I've ensured that flash is erased when attempting each method.)

ArnieO, 2022-05-10:
boot_app0.bin seems to be (only) needed if you're flashing an ESP32-S2, not sure if it makes a difference if you include it also for ESP32. 

This is the flash command line I used last time I flashed ESP32: 
esptool.py --chip esp32 --port COM3 --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 bootloader_dio_40m.bin 0x8000 ams2mqtt-esp32-partitions-2.0.11.bin 0x10000 ams2mqtt-esp32-2.0.11.bin

What are your symptoms? 
Depending on your setup you might need to press RESET after flashing is done (even if esptool.py attempts to do it for you).

mrwee, 2022-05-10:
Thanks. So following the WiKi we have identical flash setups.
Symptoms is no AP at all. Looking at serial debug output, it should start, but none of my devices can see the SSID. Have tried on two different ESP32's. Very strange... I've ordered a new/different board, to verify if I have two broken boards.

mrwee, 2022-05-10:
Just borrowed a completely new & unused board. Using the non boot_app0.bin flash method, does not bring up the AP on this board either :(. I never have issues with these AP's and usually see them on my iPhone or Android phones. It's very strange and I doubt that a brand new board is faulty.

Debug output:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12776
load:0x40080400,len:3032
entry 0x400805e4
Sensors: 255
[    66][E][esp32-hal-gpio.c:95] __pinMode(): Invalid pin selected
(I) (setup)(C1) AMS bridge started
(I) (setup)(C1) Voltage: 0.00V
(D) (setup)(C1) ESP32 LittleFS
(D) (setup)(C1)  size: 1507328
(I) (setup)(C1) No configuration, booting AP
(I) (swapWifiMode)(C1) Swapping to AP mode
(I) (setupHanPort) Setting up HAN on pin 16 with baud 2400 and parity 11
(D) (setupHanPort)(C1) Hardware serial
(D) (loop)(C1) Used 0 ms to update temperature
(D) (loop)(C1) Used 0 ms to update temperature

mrwee, 2022-05-10:
Since broadcasting an SSID isn't related to the IP stack, then that's not the issue, unfortunately.  
But I did see the SSID once or twice, but was unable to connect. Haven't seen them since. 
The ESP32's are all of the same origin, so it could point towards a faulty batch :(

ArnieO, 2022-05-10:
Maybe a dumb suggestion again, but is your power supply strong enough and stable? Consumption in AP mode is quite high.

mrwee, 2022-05-10:
No dumb questions. All hints are welcome 😀
But it is connected to a beefy PSU with good cable, but it was also my initial thought. Will try a 3.3V PSU tomorrow to see if it’s the USB connection

ArnieO, 2022-05-10:
Worth a try, the probability of finding 2-3 defective ESPs is small, 
Solid decoupling of the ESP will also do no harm.

gskjold, 2022-05-11:
Cannot say I have ever had this problem. And from the log it looks like it switches to AP mode and stays there. What kind of board is this?

mrwee, 2022-05-12:
It's "Mini D1 ESP32" with a Wroom-32 chip on it

mrwee, 2022-05-12:
Ok, this is strange. Just reflashed two modules with ESPHome, and both work flawlessly. So modules work :)
So something in my flashing with AmsToMqtt doesn't work. Any ideas anyone? The files are taken directly from the ams2mqtt-esp32-2.1.2.zip file.

mrwee, 2022-05-12:
Just tried with the GUI flasher (ESP32 Download Tool v3.9.2). Flashes fine, but no inital AP :(

[image]

ArnieO, 2022-05-12:
Very strange indeed.
Did you try erase flash before reflashing with the code? Sometimes that solves strange issues.

mrwee, 2022-05-12:
Yes. Tried that a lot of times :(

ArnieO, 2022-05-13:
> It's "Mini D1 ESP32" with a Wroom-32 chip on it

OK, I'll give it a last try. 
I'm convinced that your issue has nothing to do with the code, so that leaves us with the HW. 

These modules can be of dubious design and build quality. Even if the ESP32 does not seem to detect a brownout situation, I suspect this is a power issue. Try soldering a beefy, low-ESR (ceramic, not electrolytic) capacitor directly between Vcc and GND on the WROOM.

gskjold, 2022-05-13:
Weird indeed... I really have no good input here, but very curious what this could be. Wemos never made a ESP32 version of D1 as far as I know so I assume a copy. Do you have a link to where you can buy one?

mrwee, 2022-05-23:
Issue found after lots of troubleshooting :)
Can't recall where I bought the modules, but it's correct that they're not original Wemos.

The issue was indeed power supply. For some reason, the module worked fine with AMS2MQTT for months on USB power, but suddenly it didn't work anymore. The fact that it works with ESPhome could be much less resources used, hence less power needed (Just a theory, haven't measured). 
So I added a proper 3.3V power supply together with a 100nF + 10nF capacitors, and now it works without any issues. 
Thanks for all your suggestions and support :)

closed 2022-05-23 by mrwee

ArnieO, 2022-05-24:
Thank you for sharing!
It is not always possible to pinpoint issues related to inadequate power, as there are so many variables. When dealing with RF devices that pull huge current spikes during transmission, generous decoupling will never do any harm. And I think you're right that AmsToMqttBridge is especially demanding by doing frequent Wifi reconnection.
Glad to hear you nailed it, well done! 👍