UtilitechAS/amsreader-firmware #279: Users with an Easee car charger please help support my feature suggestion

ujhede, 2022-05-21:
Not really a problem but I would appreciate support for my suggestion on the easee upvoty.

My suggestion is that Easee implementes reading meter data form MQTT to balance load on the main fuse.

https://feedback.easee.com/b/ideas/main-fuse-loadbalancing-with-exising-smart-meter-interface-mqtt/

I appreciate your support for this suggestion.

ArnieO, 2022-05-21:
EV charger load balancing is an interesting topic, usually managed by smarthome systems in various ways.

In an ideal world, there would be a standardized MQTT payload format from energy meter readers, intended for third party devices such as EV chargers. That would certainly open many interesting possibilities.

gskjold, 2022-05-23:
Cool idea indeed. I doubt they will implement any local support for MQTT. At least my 2019 charger is ESP32 using MQTT with SSL to communicate with a cloud based MQTT. Adding another MQTT broker in the mix could be messy, so I guess they would try to run everything through their cloud solution. Would not be surprised if the communication between the equalizer and charger already do this.

With your home automation system you should be able to achieve this using https://developer.easee.cloud/docs/load-balancing ? 

Maybe HA or other have bindings for Easee?

gsolem, 2022-08-24:
I like this idea, but why can you not do this from HA or any other smart home system?

closed 2022-09-05 by gskjold