UtilitechAS/amsreader-firmware #280: Can I see exporting Watt?

HenrikDK1971, 2022-05-21:
**Describe your problem**
The payload from the MQTT if there is sun cells, I start to export Watt, is there any way I can see this?

All I can see is that it goes to 0

{
    "id": "58:BF:25:82:82:94",
    "name": "kampstrup",
    "up": 8665,
    "t": 1653129530,
    "vcc": 0,
    "rssi": -45,
    "temp": -127,
    "data": {
        "lv": "",
        "id": "",
        "type": "",
        "P": 0,
        "Q": 273,
        "PO": 1,
        "QO": 0,
        "I1": 0.25,
        "I2": 0.21,
        "I3": 0.1,
        "U1": 231,
        "U2": 230,
        "U3": 231,
        "PF": 0,
        "PF1": 0.97,
        "PF2": 0.83,
        "PF3": 0.82,
        "tPI": 38353.36,
        "tPO": 101.44,
        "tQI": 4077.94,
        "tQO": 2154.23,
        "rtc": 1653129530
    },
    "realtime": {
        "h": 0.13,
        "d": 3,
        "t": 5,
        "x": 2.84
    }
}

**Hardware information:**
 - Meter: Kampstrup
 - AMS reader: ESP32
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.1.2
 - MQTT: yes
 - HAN GPIO: [e.g. GPO12
 - Temperature sensors [e.g. 3xDS18B20]

ArnieO, 2022-05-21:
See https://github.com/gskjold/AmsToMqttBridge/wiki/Message-formats#list-4

PO is active power, export.

HenrikDK1971, 2022-05-21:
Another stupid question, what is reactive import should I add it to active import and those are together total?

ArnieO, 2022-05-22:
The short and simple answer is: Just forget about reactive power. 
What you pay for is active power.

The long answer is complex and… long. But most users have no interest in reactive power.

HenrikDK1971, 2022-05-22:
Thank tou

closed 2022-05-22 by HenrikDK1971