UtilitechAS/amsreader-firmware #281: Document how to flash with Raspberry PI

bardahlm, 2022-05-22:
After getting corrupt flash, probably because the web session was interrupted I needed to reflash the unit. I do not have a FTDI unit, so I tried to find out if I could use a Raspberry PI that I have. And it worked!

Cabling:

PI port 2 (5V) to AMS "+"
PI port 6 (Ground) to AMS "GND"

PI port 8 (GPIO UART TX) to AMS "tx" (J2)
PI port 9 (GND) to AMS "GND" (J2)
PI port 10 (GPI UART RX) to AMS "rx" (J2)

I programmed the AMS with the following command:

`esptool --chip esp8266 --port /dev/ttyAMA0 --baud 115200 --before default_reset --after hard_reset write_flash 0x0 ams2mqtt-esp8266-2.1.2.bin`

bardahlm, 2022-05-22:
Output when running:

```
esptool.py v2.8
Serial port /dev/ttyAMA0
Connecting........_____....._
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: a8:48:fa:c1:e0:76
Enabling default SPI flash mode...
Configuring flash size...
Auto-detected Flash size: 4MB
Erasing flash...
Took 2.69s to erase flash block
Wrote 732160 bytes at 0x00000000 in 80.7 seconds (72.6 kbit/s)...

Leaving...
Hard resetting via RTS pin..
```

ArnieO, 2022-05-22:
Well done, and thank you for your contribution!

gskjold, 2022-05-23:
Cool, thank you, I will add this to the Wiki soon :)

bardahlm, 2022-05-23:
I'm a bit puzzled about that I had to connect "TX" on the PI to "TX" on the AMS, shouldn't these be crossed?

ArnieO, 2022-05-23:
> I'm a bit puzzled about that I had to connect "TX" on the PI to "TX" on the AMS, shouldn't these be crossed?

If this is a Pow-U or Pow-K board, the FTDI connection point is marked with the designation of the FTDI. So the terminal marked Tx is connected to ESP Rx line and vice versa. 

We can discuss at length which marking is the most logical - but this is how I chose to do it! 😄

bardahlm, 2022-05-23:
I have a Pow-K. The wiki page then need to describe the ports on the AMS board.

ArnieO, 2022-05-23:
The firmware is intended to be run on any suitable hardware, so the Wiki is not a suitable place for that information. 

You need to consult the Pow-K User manual (See "Vedlegg" on the [product page](https://amsleser.no/hjem/1-pow-k-kamstrup.html)), chapter 3, page 8. The text could probably be written even clearer ("Connect Rx on FTDI to terminal marked Rx on the PCB" etc)- but the information is there. 😉

closed 2022-06-04 by gskjold