UtilitechAS/amsreader-firmware #282: Update to Supported Hardware in Wiki

PerJarlemark, 2022-05-25:
I can confirm that this is working on "Kamstrup DK-8660 Omnipower Three-phase meter” currently located in Sweden

**Tested Meter.**
Model: Kamstrup DK-8660 Omnipower Three-phase meter
Parity: 8N1
Baud rate: 2400

**Country**
Sweden

**Grid operator**
Boo Energi [https://www.booenergi.se/](url)
Require Encryption Key = Yes

closed 2022-06-04 by gskjold

ArnieO, 2022-10-16:
Hej @PerJarlemark - is your meter really encrypted?
I did not think encryption is used in Sweden?

PerJarlemark, 2022-10-17:
Not sure if every meeter in Sweden is, but i need the encryption key for it to work, i think the grid operator mentioned something that they whee still running the nowegian standar on my meeter 
, and that this might change in the future

ArnieO, 2022-10-17:
Thank you for confirming!

ArnieO, 2023-10-31:
@PerJarlemark 
We checked with Boo Energi, and apparently they do not encrypt:
[image]

ArnieO, 2023-10-31:
The encryption thing can a bit confusing. 
I have a suspicion that the firmware will interpret correctly non-encrypted meters even if you enter an encryption key. 
maybe try to remove it - and see what happens?

PS
I recommend upgrading your firmware to the latest version!

gskjold, 2023-10-31:
It will only use encryption key if the payload specifies that the data is encrypted

PerJarlemark, 2023-10-31:
Ok , have tried to remove it and you are right, it still runs fine so I guess Boo Energi is not using encryptions, (anymore?)

My setup is rock stable so will not update firmare right now (don´t fix if its not broken)

ArnieO, 2023-10-31:
> don´t fix if its not broken

😆