UtilitechAS/amsreader-firmware #283: Energy price in DKK is not working

tvt, 2022-06-01:
When setting up the ENTSO-E API integration with DKK as the currency, I'm getting prices very close to 0.

The issue seems to be that the parsing of the DKK to NOK is not working as intended.

The relevant debug log:
```
(I) (EntsoeApi) Retrieving EUR to NOK conversion
(D) (EntsoeApi)  url: https://data.norges-bank.no/api/data/EXR/M.EUR.NOK.SP?lastNObservations=1
(D) (EntsoeApi)Connection established
(D) (EntsoeApi)Receiving data
(D) (EntsoeApi)  got exchange rate 10.1373
(I) (EntsoeApi) Retrieving DKK to NOK conversion
(D) (EntsoeApi)  url: https://data.norges-bank.no/api/data/EXR/M.DKK.NOK.SP?lastNObservations=1
(D) (EntsoeApi)Connection established
(D) (EntsoeApi)Receiving data
(D) (EntsoeApi)  got exchange rate 136.2500
(D) (EntsoeApi) Resulting currency multiplier: 0.0744
```

The exchangerate should be closer to 1.36 - not 136.

The output from https://data.norges-bank.no/api/data/EXR/M.DKK.NOK.SP?lastNObservations=1

```
<Series FREQ="M" BASE_CUR="DKK" QUOTE_CUR="NOK" TENOR="SP" DECIMALS="2" CALCULATED="false" UNIT_MULT="2" COLLECTION="A">
<Obs TIME_PERIOD="2022-05" OBS_VALUE="136.25"/>
</Series>
```

I can see that the `UNIT_MULT="2"`, while it's `UNIT_MULT="0"` for EUR to NOK. That would be my best guess for where it goes wrong, but I don't know anything about norges-bank.no's API.

**Hardware information:**
 - AMS reader: Pow-K+

**Relevant firmware information:**
 - Version: v2.1.2

gskjold, 2022-06-03:
Weird, it does take UNIT_MULT into account (https://github.com/gskjold/AmsToMqttBridge/blob/cddb170f24313758d187d6adb25d3dd86b3eed7a/src/entsoe/DnbCurrParser.cpp#L42). Support was added for the very reason that exchange rate between nordic currencies have a 100 multiplier. Code was tested and worked at that point, maybe something has changed. I will set it up locally and check.

ArnieO, 2022-06-03:
@tvt Thank you for valuable feedback! 
While @gskjold ponders a fix, users can use the Multiplier in the config as workaround:
https://amsleser.no/blog/post/14-workaround-for-bug-in-firmware-v212

gskjold, 2022-06-04:
Either the format have changed slightly or I have reduced the buffer size on that parser. 64 bytes was too short to parse the <Series> line

closed 2022-06-04 by gskjold