UtilitechAS/amsreader-firmware #285: Empty "id" string in MQTT message breaks json parser

grubmueller, 2022-06-15:
**Describe the bug**
Once more a problem with the Austrian meters. I think about moving from ioBroker to Homeassistant. So I tried to connect the AMS reader to Homeassistent. Everything worked fine and out of the box but some values (e.g. current power) don't get a value in Homeassistant. By further looking into the MQTT messages I found out that all data of the topics "state" and "energy" show up in Homeassistent correct, only the "power" topic fails to be parsed (but the right values appear in the MQTT message). I think this is because of an empty string in the "id" field (2nd entry). As far as I know, empty strings are not allowed in json and maybe the Homeassistant json parser completely drops the message. 
Here's an example of the MQTT "power" message:
`{"lv":"EST5","id":"","type":"\253740976_A","P":2692,"Q":1466,"PO":0,"QO":0,"I1":0.00,"I2":0.00,"I3":0.00,"U1":0.00,"U2":0.00,"U3":0.00,"PF":1.00,"PF1":1.00,"PF2":1.00,"PF3":1.00}`

**To Reproduce**
In the AMS reader software "Homeassistant" is chosen under MQTT settings.
I think this issue is very specific to my meter. Maybe this field is empty because my meter doesn't send an ID- the ID field in the meter configuration also stays empty.

**Screenshots**
[image]

**Hardware information:**
 - Meter: Sagemcom (Energie Graz ist grid operator)
 - AMS reader: Custom setup based on D1 board

**Relevant firmware information:**
 - Version: 2.1.2
 - MQTT: yes, with Homeassistant option

Michael

gskjold, 2022-06-15:
First of all, what have you configured as MQTT publish topic? I know a common problem is that people use "homeassistant" here, which conflicts with internal HA events.

You are right that "id" is empty because the software could not detect the ID of your meter. But empty string is totally legal in JSON.

gskjold, 2022-06-15:
I think I see the problem now:
```"type":"\253740976_A"```
The backslash is not escaped, which I think would throw off a json parser.

grubmueller, 2022-06-15:
Hi, 
the MQTT topic is "smartmeter".
I just copied the json message into an online json parser: [JSON Parser](https://jsonformatter.org/json-parser)
And you are absolutely right, the empty string shouldn't be a problem.
The problem seems to be the backslash "type" entry (coming from the meter model).
Here the parser error:
`Parse error on line 1:
...ST5","id":"","type":"\253740976_A","P":2
-----------------------^
Expecting 'STRING', 'NUMBER', 'NULL', 'TRUE', 'FALSE', '{', '[', got 'undefined'`
After deleting the backslash the online parser works.

gskjold, 2022-06-15:
Right, I need to make some adjustments for that then :)

grubmueller, 2022-06-15:
Sorry for the wrong issue title then ;-)

gskjold, 2022-06-23:
This one should do the trick:
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/8964583/esp8266.zip)

grubmueller, 2022-06-24:
Thank you. Tested over night and everything works fine. Alls topics are valid in homeassistant now.

gskjold, 2022-06-24:
Excellent. Fix included in the latest 2.1.3 release

closed 2022-06-24 by gskjold