UtilitechAS/amsreader-firmware #287: ESP32 with 2.1.2 must be flashed  with USB to accept 2.1.3

tronde-ams, 2022-06-26:
No idea whether  it is a firmware or hardware bug, but it did not accept to load new firmware from WEB gui. I tried both the "upgrade" link and to load the file via system -> firmware.

gskjold, 2022-06-27:
Thanks for the heads up. This is probably software issue. I thought it was isolated to ESP32-S2, but most likely this applies for all ESP32 chips.

tronde-ams, 2022-06-27:
I have a feeling there is some memory corruption buried deep down here. I can't read the code and have it make sense to me, so only guessing from what I see.

Long time ago I told about very strange behavior like both my modules was about to self destruct. They are different hardware, one of them is the type not accepting full clock frequency. They behaved the same at the same time and I had to wipe the flash to reload new firmware.

Both my modules have up-time of several weeks. Both of them show correct accumulated values for monthly real time energy consumption, but they will reset the accumulated price and count from zero after a couple of weeks. I have not noticed if they do this at the same time, but both will reset long before the end of the month. The graphics appears to be OK.

About two weeks ago both showed corrupt prices from ENTSO-E. They did not show the same prices, but none was correct. I was not able to get them fetch correct prices, but they did read from the server. After a couple of days they started to show correct prices again.

*
How does the code handle corrupt data from the meter and ENTSO-E? Do you have a sanity check for all values? 

This is serial communication with a radio transmitter on the same board as the logic so a lot of interference must be expected, and we do not know if the meter itself will publish only sane data either. My meter is Aidon 3P/IT.

*
Is it possible to read all data from the flash and RAM and see if it appears to be OK while the unit is active?

manfi001, 2022-06-27:
It also applies to 8266 chips

manfi001, 2022-06-28:
I was referring to the impossibility to update to 2.1.3 OTA.  the 8266 must also be flashed with USB

gskjold, 2022-06-30:
Aha, I see. So you are unable to OTA by upload on ESP8266?

cmock, 2022-06-30:
I can confirm this also affects ESP32 in the not-S2 variant:
```
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
```

ArnieO, 2022-07-01:
I tested OTA upgrade of a Pow-U (ESP8266) from v2.1.2 (fresh install) to v2.1.3 and had no issues. Works as it should.

@manfi001 : If your issue is persistent: Could you try a factory reset first, then try OTA update again?

tronde-ams, 2022-07-02:
None of my two ESP-32 units running 2.1.3 calculate real time cost in July. They showed values after I updated in late June, but I did not notice whether they updated or not. Prices appears to be fetched correctly from ENTSO-E.
[image]

I do still have a feeling something in the code is corrupting part of the memory, but I have no idea about how to pin-point it.

gskjold, 2022-07-05:
From testing it seems I have been able to eliminate both the upgrade problem and the real time cost calculation problem.

I have not had enough time to look at data corruption. My test bench is currently in pieces, but I will get a ESP32 rolling once I get new parts to complete my rig.

Attached is a firmware with fix on cost calculation as well as a new feature that enables you to average more than one hour for the month max (where 3 seems to be what most grid operators in Norway have decided on)

NOTE that if you install this firmware and revert to a release version, it will enter AP mode and you will need to reconfigure everything.
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/9048762/esp8266.zip)
[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/9048764/esp32.zip)
[esp32s2.zip](https://github.com/gskjold/AmsToMqttBridge/files/9048765/esp32s2.zip)

tronde-ams, 2022-07-27:
Aidon 3p IT. No logging or MQTT. Only using web-gui.
Something went wrong with monthly accumulated price today or yesterday.

213 NOK is way too low in NO1. Appeared to be OK until now.

[image]

gskjold, 2022-07-28:
Interesting. Something definitely happened there. I will do a quality check on that part of the code

gskjold, 2022-07-30:
Definitely a uint16 overflow. Reducing precision on month cost from 0.01 to 1.0

gskjold, 2022-07-30:
Fixed in 2.1.4

closed 2022-08-10 by gskjold