UtilitechAS/amsreader-firmware #289: Please add version number to MQTT

dimnek, 2022-07-09:
I'd like to have the installed version number of AmsToMqttBridge available as a MQTT message.

That way I could get Home Assistent to compare the installed version to the latest released version on GitHub and add some kind of alert to my HA dashboard when the installed version gets outdated.

gskjold, 2022-08-11:
Version number is already published with the auto discovery messages, shouldn't it be possible to utilize this in HA?

dimnek, 2022-08-12:
You're absolutely correct, thanks! I had configured the MQTT payload from the reader as json, I hadn't noticed the HomeAssistant option.

When I changed the payload to HomeAssistant I got a several sensors automagically, and used this setup for the version check:

Added a mqtt sensor to get the version:
```
  - unique_id: 'amstomqtt_installed_firmware'
    name: 'AmsToMqtt Installed Firmware'
    icon: 'mdi:source-repository'
    state_topic: 'homeassistant/sensor/ams-929_P/config'
    value_template: "{{ value_json.dev.sw if value_json.dev.sw != null else states(entity_id) }}"
```
And a template binary_sensor to compare the installed version with the latest release:
```
  - unique_id: 'amstomqtt_firmware_status'
    name: 'AmsToMqtt Firmware Status'
    icon: 'mdi:source-repository'
    state: >
      {% if states('sensor.gskjold_amstomqttbridge_latest_release')[8:] != states('sensor.amstomqtt_installed_firmware') %}
        {{ true }}
      {% else %}
        {{ false }}
      {% endif %}
    device_class: update
```
The sensor gskjold_amstomqttbridge_latest_release comes from the HA GitHub integration with this repository added.

Seems like the HomeAssistant payload skips the realtime calculated data, but that is fixable on the HA side.

gskjold, 2022-08-12:
Aha, you had the JSON payload previously? Maybe it would be a good idea to add version number there as well.

dimnek, 2022-08-12:
Yes, I used the plain JSON payload. The HomeAssistant payload is kind of neat, except for the missing realtime calculated data.

closed 2022-08-23 by gskjold