UtilitechAS/amsreader-firmware #290: Increase energy price on graph from 1 to 2 decimals

ArnieO, 2022-07-29:
**Is your feature request related to a problem? Please describe.**
A lot of people will get annoyed by this, but my issue right now is that the energy price is so low that the graphical representation looks weird:
[image]
(I live in central Norway, and there are two main reasons for this ridiculous price:
1. We have had a terribly wet summer, and Norway is mainly powered by hydropower. The reservoirs in our region are currently overflowing
2. The grid capacity is insufficient to transport sufficiently surplus power out of the region. 

Back to the graph: It looks like the input numbers to the graph are rounded to two decimals, causing the graph to look like the above. If I use a multiplier (100x), I get this, effectively then showing the price in "øre" in stead of NOK:
[image]
But that would screw up the calculations - so not a good solution.

Hence my request:
**Is it possible to present data to the graph function with one more decimal, which should yield a better-looking graphical representation?** 

I don't mind if the numbers on the graph continue to show "0.0" (anyway, that is almost correct :grin:)

gskjold, 2022-07-30:
Larger labels tends to overlap, which is why two decimals are only used when there are less than 20 values in the graph. I was thinking I could rotate the label and increase the precision, but it doesn't look like this is directly supported by the framework. I agree it would be nice to fix. This is probably more often a problem in EUR when there are low prices.

tronde-ams, 2022-07-30:
Another option can be to make the price labels optional.

I find that the graphics is most useful to see if / when there is a reason to use less electricity. The actual accurate price is not the most important thing.

Without the price labels the graphics can be presented with higher resolution just as ArnieO suggests.

If possible, the max and min price can be shown as text above the graphics.

[image]

ArnieO, 2022-07-30:
> Larger labels tends to overlap, which is why two decimals are only used when there are less than 20 values in the graph. I was thinking I could rotate the label and increase the precision, but it [...]

OK, if graph resolution and labels are linked, this is probably not an easy fix. 

I don't think I've seen that graph resolution improves when there are fewer values - I will check tomorrow morning.
For reference, this is how it looks now:
[image]

ArnieO, 2022-07-30:
> Another option can be to make the price labels optional.

Good thinking - I'm not sure we need labels on each data point. I'd be willing to sacrifice labels if I could have improved resolution on the graph.

ArnieO, 2022-07-31:
v2.1.5 has exactly what I was looking for. Thank you! 😃 
[image]

closed 2022-07-31 by ArnieO

gskjold, 2022-07-31:
I had to do a emergency release due to a loading error of stored real time data. Included an update where the precision on the graph have been increased. Note that the labels themselves have not changed and will show 0.0, but the graph is better :)

gskjold, 2022-07-31:
Beautiful!

gskjold, 2022-07-31:
[image]
:sob:

ArnieO, 2022-07-31:
😬 
You shall be very happy that you will be refunded a large part of that!

To non-Norwegian readers:
Norway now has a temporary refund compensation thing. If the average price for a month is above 70 øre (around 0,07 EUR) per kWh, households will get refunded 70% of the cost that is above 70 øre/0.7 NOK/0,07 EUR. From October, compensation rate will rise from 70% to 90% - until March -23. 

This is made possible because a major part of the Norwegian hydropower production is publicly owned - and the production cost is around 10 øre / 0,01 EUR per kWh. So the Norwegian state now has unexpectedly huge incomes from the current high energy price.

ArnieO, 2022-08-01:
I confirm: The labels show two decimals when there are few values:
[image]