UtilitechAS/amsreader-firmware #291: Real time calculation don't show negative Values

evhetlan, 2022-07-29:
**Describe the bug**
Real time calculation don't show negative Values, only show 0 
See attach

[image]

gskjold, 2022-07-30:
Funny, I was thinking about this while driving yesterday, what a coincidence that it is reported later the very same day. I think the correct solution here would be to add another line for export day/hour/month. Will look into this for future release.

evhetlan, 2022-07-31:
Is It possible to get these values as export?
[image]

gskjold, 2022-08-10:
Please test attached firmware:
[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/9298494/esp32.zip)
[esp32s2.zip](https://github.com/gskjold/AmsToMqttBridge/files/9298495/esp32s2.zip)
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/9298496/esp8266.zip)

evhetlan, 2022-08-10:
Hei
Supert, mye bedre nå 🙂
Takk!

[cid:0c6506be-46df-4527-af4c-af681bcc3429]

closed 2022-08-10 by gskjold