UtilitechAS/amsreader-firmware #294: Daily power array zeroing not correct Kamstrup meter

Iridabike2, 2022-08-04:
Daily power has an hourly spike as the array needs to be zeroed for hour +1 

Code mod needed:

AmsDataStorage.cpp

utc.Hour needs to be +1:

else {
        if(debugger->isActive(RemoteDebug::DEBUG)) {
            debugger->printf("(AmsDataStorage) Last day update: %lld\n", (int64_t) day.lastMeterReadTime);
        }
        tmElements_t last;
        breakTime(day.lastMeterReadTime, last);
        for(int i = last.Hour; i < (utc.Hour + 1) ; i++) {
            if(debugger->isActive(RemoteDebug::VERBOSE)) {
                debugger->printf("(AmsDataStorage) Clearing hour: %d\n", i);
            }
            setHourImport(i, 0);
            setHourExport(i, 0);
        }
    }
[image]

gskjold, 2022-08-04:
Did you actually fix it by doing that? I am assuming you are using realtime/import/day from mqtt?

Iridabike2, 2022-08-04:
> Did you actually fix it by doing that? I am assuming you are using realtime/import/day from mqtt?

Yes it fixed it.   I have modded the hass mqtt to give me the day and month stats.

gskjold, 2022-08-10:
I am unable to reproduce your problem with my meter. Your proposed fix also removes a bar from the day graph. I have a different approach I expect to work for you. Firmware attached.
[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/9301635/esp32.zip)
[esp32s2.zip](https://github.com/gskjold/AmsToMqttBridge/files/9301637/esp32s2.zip)
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/9301638/esp8266.zip)

gskjold, 2022-08-16:
Fix included in v2.1.6

closed 2022-08-23 by gskjold