UtilitechAS/amsreader-firmware #296: Read just addresses

KJPOUNDY132, 2022-08-05:
hi i need to read, just address from arduino or esp32. do u have code for read address from electric meter with iec62056-21?

gskjold, 2022-08-10:
I'm not sure if I understand what you mean, you want to fetch a single OBIS code?

KJPOUNDY132, 2022-08-10:
i just wanna give an example i wanna read multiple  OBIS codes, but just OBIS codes without an mqtt or any othe cloud

gskjold, 2022-08-10:
In the current state of this software, it will only read known OBIS codes and present and/or publish those to MQTT with format described in wiki:
https://github.com/gskjold/AmsToMqttBridge/wiki/MQTT-configuration
https://github.com/gskjold/AmsToMqttBridge/wiki/Message-formats

An earlier request have been made to publish all OBIS codes from a payload to MQTT, but this have not been implemented yet
(https://github.com/gskjold/AmsToMqttBridge/issues/224)

KJPOUNDY132, 2022-08-10:
Thank u very much.

closed 2022-08-10 by gskjold