UtilitechAS/amsreader-firmware #297: no network called AMS2MQTT

cassio299, 2022-08-06:
After flashing I can't find the network AMS2MQTT I have tried on my Mac and on my iPhone It simply doesn't show up 

I have tried flashing several times with no result, + plug usb in and out + I have waited 20 mins for it to show up + tried the reset button on the device.... I then changed to another device no luck here either

**Hardware information:**
 - Country: Denmark
 - Meter: Kamstrup
 - Encryption enabled yes until I get into ams
 - AMS reader: ESP8266EX
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.1.5
 - MQTT: not jet
 - HAN GPIO: will be gpio16
 - HAN baud and parity: ??
 - Temperature sensors ???
 - ENTSO-E API enabled: ??

here are terminals readout:

macos@Midgaard ~ % esptool.py --chip esp8266 --port /dev/cu.usbserial-1470 --baud 115200 --before default_reset --after hard_reset write_flash 0x0 /Users/macos/Downloads/ams2mqtt-esp32-2.1.5.bin
esptool.py v4.1
Serial port /dev/cu.usbserial-1470
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 3c:71:bf:29:a0:f1
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00112fff...
Compressed 1125120 bytes to 701983...
Wrote 1125120 bytes (701983 compressed) at 0x00000000 in 68.0 seconds (effective 132.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
macos@Midgaard ~ %

read out device 2

macos@Midgaard ~ % esptool.py --chip esp8266 --port /dev/cu.usbserial-1470 --baud 115200 --before default_reset --after hard_reset write_flash 0x0 /Users/macos/Downloads/ams2mqtt-esp32-2.1.5.bin
esptool.py v4.1
Serial port /dev/cu.usbserial-1470
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: cc:50:e3:5d:1d:b5
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00112fff...
Compressed 1125120 bytes to 701983...
Wrote 1125120 bytes (701983 compressed) at 0x00000000 in 68.0 seconds (effective 132.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
macos@Midgaard ~ %

ArnieO, 2022-08-07:
I see you have edited your post. 
If issue is resolved, kindly close the thread.

cassio299, 2022-08-07:
Here by closed thanks for the AMS2MQTT software very useful

closed 2022-08-07 by cassio299