UtilitechAS/amsreader-firmware #301: Getting Data, but boundry flag missing

icinger, 2022-08-16:
I'm getting fine data from my meter, but alwasy have a boundra flag missing, so it doesn't decode

**Hardware information:**
 - Country: Austria
 - Meter: Sagemcom T210-D
 - Encryption enabled [yes]
 - AMS reader: [D1 mini]
 - M-bus adapter (TSS721ADR):

**Relevant firmware information:**
 - Version: [2.1.5]
 - MQTT: [yes]
 - HAN GPIO: [UART1]
 - HAN baud and parity: [2400 8E1]
 - Temperature sensors [none]
 - ENTSO-E API enabled: [no]

Telnet-Log:
`(V) (unwrapData) MBUS frame:
(V) 68 01 01 68  53 FF 00 01  67 DB 08 53  41 47 59 05 
(V) E7 A4 42 F1  FC DF 49 19  EE D7 7B 07  3C 62 C6 BB 
(V) 2B FC 31 BD  96 0D 8E EF  C7 DD 3C DD  2E 90 87 6F 
(V) 68 47 2F D2  41 E3 37 74  5B FB 6C 51  C1 ED AD 62 
(V) AA 30 41 BA  3D AE A5 8D  AE 2B 8E 7C  D1 DF C3 44 
(V) 31 28 3F AD  6B C6 17 FD  B5 56 22 AD  F7 8C 89 AB 
(V) 01 98 CC AB  8D 34 71 3D  CC CB 67 0A  FC F6 6F 56 
(V) D5 5B DE E4  D7 EA 56 58  39 B4 95 C8  4A DC 7E F3 
(V) FF F8 C1 56  FE AB 74 75  AE 7A 46 E4  81 7A 07 8B 
(V) 67 70 7D 53  F7 3A 32 75  F7 83 5B 51  F1 3D EB 3E 
(V) 49 19 57 9E  02 55 BC BF  B6 80 15 97  AB 50 45 CB 
(V) 68 5C 3B DD  6D FA 1F 36  ED CE 82 A1  16 A2 D0 80 
(V) C6 F3 E1 C6  3E D0 67 FE  CD FF 45 FD  7F 65 05 FF 
(V) 28 44 29 20  28 6C 6F 6F  70 29 E0 99  4E A3 AA 93 
(V) 2A 6A CA 68  01 01 68 53  FF 00 01 67  DB 08 53 41 
(V) 47 59 05 E7  A4 42 F1 FC  2F 5B 1A 99  D5 7B 1B 2D 
(V) F6 17 19 F1  6F D6 08 
(W) (printHanReadError) Boundry flag missing
(D) (AmsWebServer)Serving /index.html over http...
(E) (unwrapData) Ended up in default case while unwrapping...
`

Any hint, where to start?

ArnieO, 2022-08-16:
My first guess that this is a hardware issue, so that you are not receiving the payload correctly. 
Try to read up on old threads here to see if you find hints. (yes, I know... not so easy, but maybe search for "boundry flag missing")

gskjold, 2022-08-16:
According to our wiki (https://github.com/gskjold/AmsToMqttBridge/wiki/Known-hardware-configurations) you should configure baud and parity to 2400 8N1

icinger, 2022-08-17:
> parity to 2400 8N1

Yeah, sorry, it IS set to 8N1.

ArnieO, 2022-08-17:
> My first guess that this is a hardware issue, so that you are not receiving the payload correctly.
> Not sure about that, since i always get a payload of exactly 267 bytes.

OK, but there could still be bit errors.
- Do you get the same error on all payloads, never any payloads that are accepted?
- The TSS721 has a number of support components that must be set correctly to ensure reliable reading. Have you compared the signal into the TSS721 with the digital output? You should in particular look at the behavior during long sequences of '1' or '0'.
- Are you using the right output from the TSS721? It has an inverted and a non-inverted output. You should use the non-inverted.

gskjold, 2022-08-17:
The start of this data (68 01 01 68 53 FF 00 01 67 DB) is all correct MBUS payload start, but I cant find the stop tag (16) before the next payload starts, so it seems there are some data lost in between. Since 68 01 01 68 starts again close to the bottom of your debug, it suggests there is quite a few bytes missing, maybe a bad connection?

icinger, 2022-08-17:
> The TSS721 has a number of support components that must be set correctly

The adapter is use ist a -> (https://www.mikroe.com/m-bus-slave-click)
So i think, it schould be correctly set.
If i switch AMSreader to "inverted", it hangs at every single byte.

> Do you get the same error on all payloads
AMSreader runs since 48 hours now, and not a single payload got decrypted :(

Here are about 10 Minutes of data, the payload looks not that bad, but every single message got that error :(
(https://pastebin.com/uqUfNT3U)

I'll try another cable tomorrow.

ArnieO, 2022-08-18:
Good to hear that it works! 😄 

I speculate now, but this _could_ be (and I have to say *again*, if you search other threads here you'll see...) an other case of an ESP8266 development module with insufficient decoupling. This firmware exploits the wifi quite heavily, so solid power to the ESP is crucial. On the Pow-U and Pow-K units I sell from https://amsleser.no you will se a very beefy supercapacitor that supports the voltage to the ESP. It is an expensive component - and maybe a bit over dimensioned. But it is a solid solution that works well.

icinger, 2022-08-18:
That might be part of the problem, but i think, there is also a problem with UART1, since i have the same problem with ESP32 and UART1.

gskjold, 2022-08-19:
I know there are many people here who are using UART1 on ESP8266, so I think we would have heard if that wasn't working in general. Also, I have recently used UART1 on ESP8266 in testing at home without issues.

What ESP32 board do you have?

closed 2022-09-05 by gskjold