UtilitechAS/amsreader-firmware #303: The rtc (timestamp) seems to be one our off when using HomeAssistant MQTT-payload

dimnek, 2022-08-21:
**Describe the bug**
The rtc (timestamp) seems to be one our off when using HomeAssistant MQTT-payload.

E.g. I got this energy-mqtt-message at 2022-08-21T12:00:11+00:00 using MQTT Explorer:
`{"tPI":50560.12,"tPO":0.00,"tQI":350.83,"tQO":2944.86,"rtc":1661079600}`
When I compare with the data.json which I fetched a few minutes later:
`{"im":7360,"om":0,"mf":32,"i":1472,"e":0,"ri":0,"re":194,"ic":50560.186,"ec":0.000,"ric":350.830,"rec":2944.872,"u1":232.90,"u2":0.00,"u3":0.00,"i1":6.50,"i2":0.00,"i3":0.00,"f":0.00,"f1":0.00,"f2":0.00,"f3":0.00,"v":3.394,"r":-73,"t":-127.00,"u":1804,"m":11152,"em":1,"hm":1,"wm":1,"mm":1,"me":0,"p":null,"mt":1,"ds":1,"ea":{"x":3.8,"t":5,"h":{"u":0.07,"c":0.00,"p":0.00},"d":{"u":11.38,"c":0.00,"p":0.00},"m":{"u":343.62,"c":0.00,"p":0.00}},"c":1661083406}`
The timestamp 1661079600 from mqtt is 2022-08-21T11:00:00+00:00, which seems to be one hour off.
The timestamp 1661083406 from data.json is 2022-08-21T12:03:26+00:00, which is correct.
The time in the web interface displays the time correct (in my local time, UTC+2).
The values for tPI and ic are almost the same and corresponds to a manual reading of the meter.

**To Reproduce**
Steps to reproduce the behavior:
1. Wait for an energy mqtt message
2. Compare with a data.json

**Expected behavior**
I expected the rtc (timestamp) value to be close to current time.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware information:**
 - Country: Norway
 - Meter: Aidon 6515
 - Encryption enabled: no
 - AMS reader: Pow-U
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.1.16
 - MQTT: yes
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: [e.g. 2400 8E1]
 - Temperature sensors [e.g. 3xDS18B20]
 - ENTSO-E API enabled: no

**Additional context**
Add any other context about the problem here.

gskjold, 2022-08-23:
"rtc" is meter timestamp (received from meter), while "c" in data.json is clock. The data from the different manufacturers does not send data/time in a consistent format, so there are some assumptions made in the code. I have a Aidon meter myself, so I will have a look at the data around the hour to see what is going on.

gskjold, 2022-08-23:
At 21:00 the meter reports 20:00, which is not correct by any standard... I will subtract 1hr to reach UTC time which will fix the issue now, but I am not sure what happens when we switch from CEST to CET this winter...

closed 2022-08-23 by gskjold