UtilitechAS/amsreader-firmware #304: Reconnect issue mqtt

1968-Larry, 2022-08-25:
**Describe the bug**
Problem when network goes down // reconnecting to mqtt-server.

**Relevant firmware information:**
ESP32, mqtt with ssl, version 2.1.2, HAN baud 2400 8e1, entso-e api disabled.

**Additional context**
When mqtt-server is not responding for a while, the esp32 becomes unavailible. Needs power reset to come back online. Trying to reproduce without SSL with no success so far.

gskjold, 2022-08-25:
Most likely a memory leak when reconnecting with SSL, I will have a look when I got some time to spare

1968-Larry, 2022-08-25:
Without SSL it seems stable, but changing port number from working mqtt-ssl to something random makes everything unstable. Ping request works but not web.

1968-Larry, 2022-08-25:
Tried latest version 2.1.7 connected to serial debug without HAN connected.

10:11:09.240 -> [0mGuru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
10:11:10.173 -> 
10:11:10.173 -> Core  1 register dump:
10:11:10.173 -> PC      : 0x4018ad7a  PS      : 0x00060130  A0      : 0x800dba02  A1      : 0x3ffb23b0  
10:11:10.173 -> A2      : 0x3ffc2f74  A3      : 0x00000000  A4      : 0x3ffb23e4  A5      : 0x00000001  
10:11:10.173 -> A6      : 0x00000000  A7      : 0x3ffb204c  A8      : 0x00000000  A9      : 0x00000008  
10:11:10.173 -> A10     : 0x3ffb23e4  A11     : 0x00000000  A12     : 0x3ffc47a4  A13     : 0xa18e70c8  
10:11:10.173 -> A14     : 0x00000001  A15     : 0x00002000  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
10:11:10.220 -> EXCVADDR: 0x00000000  LBEG    : 0x4018ad76  LEND    : 0x4018ad85  LCOUNT  : 0x00000007  
10:11:10.220 -> 
10:11:10.220 -> 
10:11:10.220 -> Backtrace:0x4018ad77:0x3ffb23b00x400db9ff:0x3ffb23d0 0x400da09f:0x3ffb2410 0x401082a1:0x3ffb2820 
10:11:10.220 -> 
10:11:10.220 -> 
10:11:10.220 -> 
10:11:10.220 -> 
10:11:10.220 -> ELF file SHA256: 0000000000000000
10:11:10.220 ->

gskjold, 2022-08-25:
When do you get this error?

1968-Larry, 2022-08-25:
First I connect to working mqtt-ssl, change port to something random. More stable I think if booted with wrong port.

gskjold, 2022-09-27:
Found out why this happens, will be fixed in next release.

closed 2022-09-27 by gskjold