UtilitechAS/amsreader-firmware #305: Update not working on webinterface?

KlavsKlavsen, 2022-08-27:
**Describe the bug**
on the UI, I press 'upgrade' - and nothing seems to happen.

**To Reproduce**
Click 'upgrade' in webUI.

**Relevant firmware information:**
 - Version: 2.1.1
 - MQTT: yes

**Additional context**
Its the PowK model - bought via webshop a few months ago - in Denmark.

What can I do to debug, or upgrade in different way?

gskjold, 2022-08-27:
DHCP or static IP? If static, DNS needs to be configured. Can also be upgraded by upload via firmware menu

closed 2022-09-05 by gskjold

KlavsKlavsen, 2022-09-12:
DHCP. DNS is set to 1.1.1.2 - and works for me - a wifi client on same network.
Under firmware I only see 2 warnings (for uploading update myself) :(

1. it says its running the esp32s2 hw - does that mean this is the correct file to try to upload: https://github.com/gskjold/AmsToMqttBridge/releases/download/v2.1.7/ams2mqtt-esp32s2-2.1.7.bin

2. It says Units powered over M-bus must be connected to an external power supply during firmware upload. Failure to do so may cause power-down during upload resulting in non-functioning unit. 

Is the PowK model 'attached to Mbus' - when running in my electical meter as in normal operations? so I need to pull it out, and attach microusb power-supply to it, to update?

I tried enabling "telnet debugging" - hoping that would enable me to run a dns lookup on the device, to confirm it works - but that was not an option.

Hitting 'upgrade' - which I would hope would mean I do not need to worry about the 2 warnings - still does nothing.

gskjold, 2022-09-12:
Pow-K does not need external power. And you have the correct URL for firmware for your board.

I cant remember the details now, but for v2.1.3, a bug regarding S2 OTA firmware upgrade was fixed. So upgrade by uploading the file this time, and next time the upgrade via GUI should work.

KlavsKlavsen, 2022-09-12:
I would propose to improve the warning to state that PowK has enough power.. just to help with worrying :)

gskjold, 2022-09-12:
Yes, this is in the works, but since Pow-U and Pow-K has the same profile, it is currently not able to detect the difference.

KlavsKlavsen, 2022-09-12:
thank you for your help @gskjold I got it updated to 2.1.7 now :)