UtilitechAS/amsreader-firmware #306: 2.1.7 / ESP32 / Aidon 3p IT doers not publish data for L3 in MQTT raw full

tronde-ams, 2022-08-28:
ESP32 / Aidon 3p IT Norway

No logging.
I have not used MQTT for long time, so I don't from which version this appeared first.
I am using public broker broker.emqx.io without SSL.

{publish topic}/meter/l3/current
{publish topic}/meter/l3/voltage

is not published for MQTT raw full.
MQTT raw minimal does publish them as expected.

My meter does not publish power factor, so I don't kow if it is affected as well.

gskjold, 2022-09-25:
Could you try this version:
[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/9640340/esp32.zip)

tronde-ams, 2022-09-25:
feb8e50 behaves the same as 2.1.7 :-(

I use MQTT Explorer to monitor.

It also seems like there are some kind of bug related to prices. I have noticed it in 2.1.7, but not paid much attention at the moment. Will open a new issue if I get a better impression of what it is about., but maybe you should look at the retain flag for prices? 

**This is RAW full** (time 15:14:49). RAW full was activated several minutes before the screen shot.
No data for L3

[image]

**This is RAW minimal** (time 15:15:55)
OK data for L3

[image]

gskjold, 2022-09-27:
Currently debugging with RAW full and consistently receiving data for l3. Maybe you are hitting some limits of emqx? https://docs.emqx.com/en/cloud/latest/intro/restriction.html#connected-communications

tronde-ams, 2022-09-27:
Maybe? But I find it strange that it always affect L3-

I will try some of the other public brokers an see what happens.

tronde-ams, 2022-09-27:
I have changed to test.mosquitto.org, and everything including prices seems to be OK. I use the same hardware running https://github.com/gskjold/AmsToMqttBridge/commit/feb8e5007bb210557c74cacfcd2699ad836595a8 so it seems like EMQX was the culprit here.

gskjold, 2022-09-28:
Good to know!

closed 2022-09-28 by gskjold