UtilitechAS/amsreader-firmware #307: Cannot store a new value for average of top x hours

stigvi, 2022-08-28:
**Describe the bug**
Saving a new value for "Average of top X hours" has no effect. It is always 1 hour.

**Hardware information:**
 - Country: Norway
 - Meter: Aidon
 - Encryption enabled no
 - AMS reader: ESP32

**Relevant firmware information:**
 - Version: 2.1.7
 - MQTT: yes

tronde-ams, 2022-09-22:
ESP32 / Aidon 3p IT / only using WEB GUI.

Something is wrong with this average function. Mine is set to average top 3 hours. This was carried over from previous version when I upgraded to 2.1.7, and I have not tried to change it. It does not average, but consistently shows the maximum hourly value, even if it appears within same day.

ArnieO, 2022-09-23:
I believe the averaging is implemented according to the new "Energiledd / Kapasitetsledd" implemented in Norway this summer. 
The pricing model is averaging **3 highest hourly consumption on different days** ("gjennomsnittet av de tre høyeste døgnmaksene i måneden"). So I believe what you observe is a correct implementation of this model. 
You will have to wait for a new "high hour" on a different day to see change on the averaged value.

tronde-ams, 2022-09-23:
I know, but my impression is it does not.  
I will wait and see what happens next month when I will get larger peaks. First three days should be "wrong", but after that it should show correct average if OK.

ArnieO, 2022-09-23:
> I know, but my impression is it does not. I will wait and see what happens next month when I will get larger peaks. First three days should be "wrong", but after that it should show correct average if OK.

My impression is that this works fine - but I might not have followed the evolution of the average value closely enough.
This feature is relatively new, so let us (all?) follow it a bit closer start of next month and see if we detect a snag.

gskjold, 2022-09-25:
Number of top hours was saved, but was only applied by reboot. Fixed.

As for the calculation, mine seems to be correct at least. We should monitor this to be certain though

closed 2022-09-25 by gskjold