UtilitechAS/amsreader-firmware #308: Can not  enter meter type and model

Duckshousemacservice, 2022-08-31:
**Describe your problem**
Can not enter meter type and model

the fields are grayed ut

**Hardware information:**
 - Country: danmerk
 - Meter: kamstrup
 - Encryption enabled yes
 - AMS reader: ESP32 

**Relevant firmware information:**
 - Version:  2.1.7
 - MQTT: yes
 - HAN GPIO: GPIO3
 - HAN baud and parity:  2400 8E1

 - ENTSO-E API enabled: no

gskjold, 2022-08-31:
These are no longer user definable fields. On most meters this will be autodetected, but on some it remains blank. If you truly have 2400 8E1 like you have specified above, change to parity 8N1 which is correct for your meter. More in wiki: https://github.com/gskjold/AmsToMqttBridge/wiki/Known-hardware-configurations

Duckshousemacservice, 2022-08-31:
i did have 8E1.  sorry.   but i still cant connect. i tried all 3 uarts.  with 8n1...  ill contact my provider.

gskjold, 2022-08-31:
How have you connected the HAN port to your hardware?

Duckshousemacservice, 2022-08-31:
a header on 2 wires.  1 to ground and 1 til gpio 16.  first pin i han is ground and second is tx data out. as i understand...

gskjold, 2022-08-31:
Then UART2 is your friend. I am not sure of the Kamstrup pinout, but I had this image on disk that may or may not be correct:
[image]

Duckshousemacservice, 2022-08-31:
it is correct according to the info i got from my provider Andel A/S. ill try uart2 and if its not working ill grab a new board. i might killed it during research....  thanks for the quick answering!

ArnieO, 2022-08-31:
> I had this image on disk that may or may not be correct:
> [image]

It is correct.
https://github.com/ArnieO/SmartMeterDocumentation#han-interface-description

Duckshousemacservice, 2022-08-31:
got it now.  new esp did the trick. thanks a lot. now off to tinker more...

closed 2022-08-31 by gskjold