UtilitechAS/amsreader-firmware #309: Expose Peaks etc in Home Assistant mode

roflmao, 2022-09-04:
**Is your feature request related to a problem? Please describe.**
Raw MQTT has more functionality than Home Assistant mode, expose more or everything raw so it can be used manually in HA in addition to Discovery.

**Describe the solution you'd like**
Expose all raw MQTT with Home Assistant mode.

**Describe alternatives you've considered**
MQTT Raw all the things, but still nice to have autoconfig with MQTT Discovery.

**Additional context**
Great project! Thanks alot.

gskjold, 2022-09-28:
Work in progress. Currently implemented auto configuration for sensors containing price and realtime data.

What I need to verify:
- The new firmware functions with device configured from old auto config data (Version 2.1.7 or below)
- The new price sensors are functioning as expected (Has future price 0-11, min/max price ahead and 1hr, 3hr and 6hr cheapest times ahead)
- Real time sensors functioning as expected (Has month max, threshold, peaks, hour/day/month use/cost/produced)

WARNING: Use the DEV firmware below at your own risk. You need to be able to flash by USB/TTL if firmware does not work.
[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/9667912/esp32.zip)
[esp32s2.zip](https://github.com/gskjold/AmsToMqttBridge/files/9667913/esp32s2.zip)
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/9667914/esp8266.zip)

gskjold, 2022-09-30:
Merged this issue into #320

closed 2022-09-30 by gskjold