UtilitechAS/amsreader-firmware #310: last hour usage

kitten77, 2022-09-07:
Would it be possible to include the last hour of usage as a variable in the data.json set?

gskjold, 2022-09-30:
This information is already available in dayplot.json where usage for the last 24hrs are stored by UTC hour. Having a "usage last hour" variable would be convenient, but it will also consume more memory (although a very small amount) which I am very careful with these days. Memory is best reserved for added features.

closed 2022-09-30 by gskjold

kitten77, 2022-12-16:
That i can agree with you on, memory is a precious thing.

Been looking at the dayplot.json and i am not sure how i should read this. 
Today is 16.12 07:00 that would make 
i00 15.12 00:00 ? 

Should the e00 be 16.12 00:00 cause that one reads 0. 

Also been looking for the jsonprices.json which won't load in the browser for some reason, is that only being posted to the mqtt?

gskjold, 2022-12-16:
i00 through i23 is import in UTC time for the last 24 hours. So i00 right now would be 16.01 00:00 UTC or 01:00 CET. 

For prices, the URL would be /energyprice.json