UtilitechAS/amsreader-firmware #311: MQTT Home Assistant payload

drzero42, 2022-09-10:
**Describe the bug**
I have AMS setup to publish the full payload to an MQTT topic. After upgrading to 2.1.7, I have tried switching to the Home Assistant payload type. But after I do that, the MQTT box at the top becomes red, and at the bottom of the dashboard it reports "MQTT Communication Error -1". It looks like data is published to the topic set, but nothing related to AMS shows up under the `homeassistant/sensor` topic, as mentioned in the documentation. If I remove the publish topic (set the field to empty), the MQTT box becomes green and the error goes away, but nothing seems to get published to MQTT at all.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to MQTT configuration
2. Setup to publish to a specified topic
3. Set payload to Home Assistant
4. Error is given on dashboard

**Expected behavior**
Data is published to configured MQTT topic and auto discovery information for Home Assistant is published to the topic `homeassistant/sensor`.

**Hardware information:**
 - Country: Denmark
 - Meter: Kamstrup
 - Encryption enabled: yes
 - AMS reader: ESP8266

**Relevant firmware information:**
 - Version: 2.1.7
 - MQTT: yes

jorgror, 2022-09-10:
I had a similar issue. I can't recall the exact procedure I used to fix it, but you should try rebooting the ESP and changing back and forth on the payload.

drzero42, 2022-09-10:
Well, what do you know. Changing the payload, keeping the topic and then rebooting the ESP did the trick. Thank you! :)

closed 2022-09-10 by drzero42