UtilitechAS/amsreader-firmware #314: Negative current shown as positive current.

rndfm, 2022-09-18:
When exporting to the grid one or more phases must have negative current.

But all phases is shown with positive current?

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Export power on one or more phases
2. Observe no phases is shown with a negative current.

**Expected behavior**
Phases should be negative when power is exported on that phase.

**Hardware information:**
 - Country: Denmark
 - Meter: Kamstrup 3P
 - Encryption enabled: yes
 - AMS reader: POW-K+, HAN-PORT LESER FOR KAMSTRUP STRØMMÅLER

**Relevant firmware information:**
 - Version: 2.1.7
 - MQTT:  yes
 - HAN GPIO: 
 - HAN baud and parity: 2400 8N1

ArnieO, 2022-09-18:
There is no sign on the current data coming from the meter, so no negative current reported. Power is reported as either net Import or net Export, this is how the direction of energy flow is reported. 
So what you see is normal behavior.

rndfm, 2022-09-20:
I'm sorry the hear that but thanks for clarifying.

I will go buy me another meter :)

closed 2022-09-20 by rndfm