UtilitechAS/amsreader-firmware #316: No ENTSO-E data published on MQTT

tb205gti, 2022-09-22:
**Describe your problem**

ENTSO-E values not being pushed on MQTT.

I have selected the json payload, but I do not get any ENTSO-E data published in the json structure, nor can I find any description of what it is supposed to look like on the wiki. Only for raw format. If I select RAW I still do not get any pricing data published on MQTT. MQTT is working fine for meter data.

**Hardware information:**
 - Country: Denmark
 - Meter: Kamstrup
 - Encryption enabled yes
 - AMS reader: POW-K+
 - M-bus adapter (if applicable): none

**Relevant firmware information:**
 - Version: 2.1.7
 - MQTT: yes
 - HAN GPIO: GPIO16
 - HAN baud and parity: 2400 8N1
 - Temperature sensors : none
 - ENTSO-E API enabled: yes

ArnieO, 2022-09-23:
I believe the pricing payload is currently only generated if the Home Assistant MQTT payload format is selected.

gskjold, 2022-09-23:
Actually it is the other way around, only JSON and RAW has prices. It should be published in separate JSON when prices are updated once per day.

Format shown here: https://github.com/gskjold/AmsToMqttBridge/wiki/Message-formats#prices

tb205gti, 2022-09-23:
So the topic is the same - but with a different payload? or is the price structure appended to the normal json object?

gskjold, 2022-09-23:
Same topic, different payload with format as show in previous link

gskjold, 2022-09-25:
Was this resolved? Tested locally and seems to work fine here

dalpil, 2022-09-25:
Works just fine here as well (version 2.1.6), the prices gets transmitted as JSON via MQTT at the start of every hour.

Maybe mention the fact that the prices only get sent at the start of every hour on the wiki page for [MQTT Configuration](https://github.com/gskjold/AmsToMqttBridge/wiki/MQTT-configuration) (or on some other page)? I couldn't find it documented anywhere.

tb205gti, 2022-09-25:
I think it is a good idea to document it - right now it is not documented and I was going mad trying to figure out why i did not get any data :)

gskjold, 2022-09-27:
Added to wiki https://github.com/gskjold/AmsToMqttBridge/wiki/MQTT-configuration#json-mode

closed 2022-09-27 by gskjold