UtilitechAS/amsreader-firmware #317: Per phase usage

Andreasjkoch, 2022-09-27:
**Describe your problem**
Is it possible to read out the current per phase usage in watts? 

**Hardware information:**
 - Country: Denmark
 - Meter: Kamstrup
 - Encryption yes
 - AMS reader: Pow-K+

**Relevant firmware information:**
 - Version: 2.1.7
 - MQTT: yes

ArnieO, 2022-09-27:
No, the firmware only provides the values coming from the meter. 
There are some select cases where some types of meters do not provide a parameter because it can be calculated from others. In those cases the firmware does the calculation and "fills in the blanks". 

The meter does not provide power or energy per phase, so the firmware does not report it.

Andreasjkoch, 2022-09-27:
Can it be calculated then? I have to be honest, I do not know much about this topic, but from reading online the wattage per phase should be able to be calculated with I * V * PF. I do get power,voltage and powerfactor for each phase, but when trying to do this calculation it does not add up to the correct total.

ArnieO, 2022-09-27:
To be honest, it does not make much sense to try to calculate power per phase - which is why no meter reports it. Current change in one phase gives current change in other phase(s).
The reason your calculation does not add up is that you need to take into account that the phases are 120 degrees phase shifted, so you need to do vector calculations.

tronde-ams, 2022-09-27:
Given you have 400 V TN distribution system (I guess you have since in Denmark)  I * V * PF for each phase is correct. 

BUT -  and this is a very large BUT:  The accumulated value for energy (kWh) is calculated based on probably several thousand calculations every second, while the values for voltage, current, watt and PF is an averaged snapshot (probably only 0.5 sec. - 1 sec.) pushed from the meter every 10 second. 

The implication of this is that the accumulated value for energy is as correct as it can be, while the values for voltage, current, watt and PF can be quite "wrong".  

We must remember that the accumulated energy consumption is the main priority for for this meter while the other values are more or less "nice to have".  They give the grid company a lot of useful information about the grid, but they do not need the same accuracy as for the energy you pay for. The values are also normally good enough to be used in a home automation system for load balancing. 

You can calculate as you suggest, but you must consider the result as "almost correct"  instead of correct. It can still give useful 
information.

*

If you had been on the Norwegian 230 V IT distribution system, you would have been completely off in some instances without knowing when.

gskjold, 2022-09-27:
The encrypted Danish Kamstrup actually provides active import and export per phase in the payload, but it is not extracted yet. Will be fixed by #224

closed 2022-09-27 by gskjold