UtilitechAS/amsreader-firmware #318: Make {publish topic}/meter/id  optional?

tronde-ams, 2022-09-27:
Not the biggest issue.

I have played a little bit with some of the android apps for MQTT and the public broker at broker.emqx.io and found them useful.

I know it is possible to encrypt, but it does not seem to be straightforward for noobs like me? The only unique identifier in the MQTT-stream seems to be the meter-id, and I wonder if it is possible to make it optional? I don't feel it a security risk for me to publish to a public broker with this stripped from the stream, but I can be wrong...

gskjold, 2022-09-27:
I'm sorry to say I won't implement this. Mostly because it is not something that many people need, but also because I'm trying to avoid increasing memory footprint.

closed 2022-09-27 by gskjold

tronde-ams, 2022-09-27:
No problem :-)