UtilitechAS/amsreader-firmware #321: Boot error

kaareseras, 2022-10-01:
**Describe the bug**
Esp32 Cam board never start up
error on serial out: Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

**To Reproduce**
Steps to reproduce the behavior:
Instal on ESP32-s
https://www.aliexpress.com/item/1005003334474586.html?spm=a2g0o.productlist.main.1.421b6d7cCEVAV9&algo_pvid=43691a32-f5c1-4f61-8bdd-c5c4f8df3f04&aem_p4p_detail=202210011431157264895227982480006628901&algo_exp_id=43691a32-f5c1-4f61-8bdd-c5c4f8df3f04-0&pdp_ext_f=%7B%22sku_id%22%3A%2212000025271323942%22%7D&pdp_npi=2%40dis%21DKK%2155.78%2144.62%21%21%2116.77%21%21%40214528be16646598754658228d0738%2112000025271323942%21sea&curPageLogUid=xNjz2uWyFuym&ad_pvid=202210011431157264895227982480006628901_1&ad_pvid=202210011431157264895227982480006628901_1

**Expected behavior**
Boot into AP-mode

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware information:**
 - Country: [e.g. Norway]
 - Meter: [e.g. Aidon]
 - Encryption enabled [yes/no]
 - AMS reader: [e.g. Pow-U, ESP32 etc]
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: [e.g. 2.1.0]
 - MQTT: [yes/no]
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: [e.g. 2400 8E1]
 - Temperature sensors [e.g. 3xDS18B20]
 - ENTSO-E API enabled: [yes/no]

**Additional context**
Add any other context about the problem here.

gskjold, 2022-10-02:
How did you flash your board? What version are you flashing?

If you have had some other program flashed to it earlier, please run erase_flash before flashing.

kaareseras, 2022-10-02:
I used Esptool as descibed in wiki. Exact same command. The only diff is the type of mcu; Esp32-s. I have installed it fine and working on Esp32 wroom. Maybe another file is needed for the bootloader?

gskjold, 2022-10-02:
Could be, I am not familiar with that specific packaging.

kaareseras, 2022-10-02:
I tried this device, as its the only I can find with external antenna. I need that as my meter is inside a steel cabinet and signal is not stable with internal pcb antenna (-84db). I have hacked the antenna onto a wroom board, and it works fine (-55db). However this bug relates to the package of esp32-s and I believe this package relevant due to the external antenna. Link to the device is in the bug description

gskjold, 2022-10-02:
I saw  your link, but what I really need is a proper spec on the packaging. Espressif does not have a ESP32-S listed as one of their SoC's, so this is some kind of third replica packaging. Ref: https://www.espressif.com/en/products/socs