UtilitechAS/amsreader-firmware #323: Very small bug in gui

drlauridsen, 2022-10-08:
Minor bug in GUI.

When choosing any page, the ENTSO-E API link disappears from the menu.
The pagelink is shown only from the main page, and can be shown again simply by going back to the main page.

[image]
[image]

ArnieO, 2022-10-09:
Yeah - I've seen this too, but have forgotten to report it. Thank you!

gskjold, 2022-10-12:
Sorry I have also seen this, but have ignored it.. Will improve :)

gskjold, 2022-11-21:
Fixed in upcoming version 2.2.0

gskjold, 2022-12-06:
v2.2 beta out for testing now:
https://github.com/gskjold/AmsToMqttBridge/discussions/370