UtilitechAS/amsreader-firmware #324: error msg on Failed to connect to MQTT

CHHE-ledtec, 2022-10-11:
**Describe your problem**
A clear and concise description of what the problem is.

**Hardware information:**
 - Country: sweden
 - Meter: --
 - Encryption enabled [yes/no]
 - AMS reader: [Pow-k+]

**Relevant firmware information:**
 - Version: 2.1.8
 - MQTT: yes . hivemq cloude ( free account)
 - ssl: yes
 - have cert ( need better wiki about this subject )
 
error is that i cant connect with mqtt and only error msg i get is 

Failed to connect to MQTT: -3

-3 does not say anything to me or google.

allso to little info in wiki and on the mqtt config aboute ssl what file to put where.

as it states now i says.

CA  upload  - Certificate upload - private key upload

and i have 2 certifaction i have made. not 3 only 2
and thos are 

client-cert.crt
server.crt

i have 3 pem files if it is them i am ment to us e

client-cert.pem
client-key.pem
server.pem

gskjold, 2022-10-12:
Error code 3 has no description other than it failed to connect.

Upload server.pem to CA, client-cert.pem to Certificate and client-key.pem to private key.

Agree that better documentation is needed! I will keep this issue open until i fix this

gskjold, 2022-10-15:
This is not the correct list of error codes. The error codes come from a enum in lwmqtt:
https://github.com/256dpi/arduino-mqtt/blob/69cb7d2111f9eed92b74ccd57f128b4e1e989934/src/lwmqtt/lwmqtt.h#L15

closed 2022-11-01 by gskjold