UtilitechAS/amsreader-firmware #325: ESP32 not powering on from the meter

Gameroad, 2022-10-12:
**Describe your problem**
Not being able to power my ESP32 from the M-Bus module.
So this is probably a hardware fault, but I taught I might ask this question here since the documentation on these modules from China is zero to none.
I have connected my ESP to the module as described in the WIKI, but my ESP does not power up from the meter. When I connect a power bank to the ESP everything works and I get data from the meter.
Since I have only tried it with a power bank I haven't been able to verify all the data but it looks great so far.
I have tried to bridge the jumper on the module but that does nothing. 
Have anyone had this issue before? 

**Hardware information:**
 - Country: Sweden
 - Meter: Aidon 6534
 - Encryption enabled: No
 - AMS reader: ESP32
 - M-bus adapter (if applicable): TSS721, same as the one in the WIKI
 
[image]
[image]

ArnieO, 2022-10-12:
1. How are you trying to power the ESP32? The red wire in the photos is connected to a terminal named V_in, I expect this terminal is for powering the electronics on this converter board - and not a voltage output. 
2. The TSS721 M-bus transceiver chip used in this module *does* have a voltage output, but it delivers very little power (find and read its datasheet). It cannot power an ESP32. You'll need a solution with a DC/DC converter that does not interfere with the M-bus data signal. And I doubt if it is the voltage output pin of the TSS721 that is on the Vin terminal of the board.
3. Even if you find a way to pull power from the M-bus, a "normal" ESP32 consumes too much. You need to use an ESP32-S2, and enough capacitance to act as charge buffer while the ESP is doing Wifi.

I have developed the [Pow-U](https://amsleser.no/hjem/7-pow-u-universal.html), which is powered from the M-bus. Solving the powering from M-bus without distorting the data signal was not straightforward. 😉

Gameroad, 2022-10-12:
Thank you!
I totally misinterpreted the Wiki, now I understand I need to power the ESP32. Everything is working great 👍 

I have seen the unit you have developed, they look great 👍 good job!

closed 2022-10-12 by Gameroad