UtilitechAS/amsreader-firmware #327: Throttle how often the real-time sensors update

Mithrox, 2022-10-13:
**Is your feature request related to a problem? Please describe.**
After changing to AMS leser from Tibber Pulse I notice my energy graphs and statistics have become very slow compared. I am wondering if this may be because the update frequency of the AMS leser seems to be very short. I do believe Tibber had a 5 or 10 seconds refresh, so any graphs could draw correctly while it was static.

**Describe the solution you'd like**
I would like a way to throttle/control the refresh time on the sensor, so maybe instead of every 2 seconds (or it may be 1 second), we could set 5 or 10 seconds as a refresh on the sensor (the time between each message goes on the MQTT bus).

**Describe alternatives you've considered**
I have considered using Node Red to write the value of the real-time sensors to a template sensor which only update each 10 seconds, but it creates more complexity that may be avoided by controlling the rate on the MQTT bus.

**Additional context**
No additonal context from me, I welcome any opposite views or experiences as this is just my view on it.

gskjold, 2022-10-13:
This only has value for a very small group of people, so I'm sorry to say that I will not implement such feature.

closed 2022-10-13 by gskjold

ArnieO, 2022-10-13:
Data is sent to MQTT broker as the same interval as they are pushed from the meter to the AMS leser. The payload interval is:

- Kamstrup: 10 seconds
- Aidon and Kaifa: 2,5 seconds