UtilitechAS/amsreader-firmware #328: Proposal for a new display of the month's three highest consumptions

eireit, 2022-10-13:
Today only the average of the month's highest consumption per day is shown.
Suggest that you display this in table format and show which days have the highest consumption. Attached display used by the Oss app as an example.

[image]

ArnieO, 2022-10-14:
Thank you for interesting proposition!

eireit, 2022-10-15:
Do you use the average of the five highest per month?  ea":{"x":4.9,"p":[5.62,4.49,4.72,5.16,4.61] . Is the average of the three highest (on three different days) that counts. This is relevant for Lyse and Fagne (Haugaland kraft) at least.

ArnieO, 2022-10-15:
> Is the average of the three highest (on three different days) that counts. This is relevant for Lyse and Fagne (Haugaland kraft) at least.

This is what is calculated and displayed at the right of the "Real time consumption" line.
The number of hours to average is set in here: Configuration / Meter / Configure tariff thresholds / Average of top

My setting:
[image]
My current value on the "Real time consumption" line:
[image]
This means: My current average of the top 3 hours in 3 different days is 6.7, and the nest threshold level (to be avoided) is 10.

eireit, 2022-10-15:
Thanks for the explanation.

gskjold, 2022-11-23:
A visual presentation of peaks will be included in v2.2, but it will not be as detailed as this.

closed 2022-11-23 by gskjold