UtilitechAS/amsreader-firmware #329: Add electricity price hour by hour in json format

eireit, 2022-10-15:
Is it possible to add the electricity prices as hour by hour in json format.
In http://<ams_hostname>/data.json or a separate http://<ams_hostname>/prices.json

price_today_00-01
price_today_01-02
price_today_02-03
price_today_03-04
price_today_05-06
price_today_06-07
price_today_07-08
price_today_08-09
price_today_10-11
price_today_11-12
price_today_13-14
price_today_15-16
price_today_16-17
price_today_17-18
price_today_18-19
price_today_19-20
price_today_20-21
price_today_21-22
price_today_22-23
price_today_23-00
price_today_Average
price_today_Max
price_today_Min
price_today_current

gskjold, 2022-10-15:
Already at /energyprice.json

closed 2022-10-15 by gskjold

ArnieO, 2022-10-15:
Maybe the various *.json possibilities should be listed in the Wiki?

eireit, 2022-10-15:
yes please....