UtilitechAS/amsreader-firmware #330: Lost connection after upgrade to 2.1.9

rtollevik, 2022-10-16:
I upgraded my Pow-K+ from amsleser.no to v2.1.9, and lost connection with the reader :-( Apparently, it failed to reconnect to my wifi network.  Alternating green and yellow flashes on the led.  I was able to reset the device and connect using my mobile as access point.  After downgrade to v2.1.8 it connected to my ordinary wifi network.  I don't known whether it is the longer ssid name and password on my ordinariy network which is the problem, or something else with that network. I'll experiment more tomorrow and update with what I learn.

gskjold, 2022-10-17:
Wow, this is a pretty serious problem, I'd like to know if there are more people who experience this in case I have to remove this version.

ArnieO, 2022-10-17:
I updated now, it restarted as it should. 
[image]

ArnieO, 2022-10-17:
> I upgraded my Pow-K+ from amsleser.no to v2.1.9, and lost connection with the reader :-( Apparently, it failed to reconnect to my wifi network. Alternating green and yellow flashes on the led. I was [...]

Yes, it would be great if you could do some experiments, and be very precise on each step, in case you manage to recreate same or similar issue. Did you use one-click update, or did you download files and update from them? If so, are you sure you used the right files? (ESP32 files will not work on Pow-K+, you need ESP32S2 files). 

Alternating green and yellow flashes sounds very peculiar.

gskjold, 2022-10-17:
OK, I have deleted 2.1.9 now. I will investigate what may cause this problem.

rtollevik, 2022-10-17:
I upgraded using the GUI link.  It appear to be something with my wifi, the same SSID and password works when I use my mobile as access point.  Any way I can access the serial debugger on a mac?
A video showing the flashes is attached.

https://user-images.githubusercontent.com/17461471/196131605-099615eb-369d-4347-9c64-6eaeb966fbec.mp4

gskjold, 2022-10-17:
This is a reboot loop, wonder what that could be... Anything special configured, or does this happen for å "blank" configuration just after initial config?

henrikkoch, 2022-10-17:
> Yes, me too. I spent an hour yesterday to get 2.1.9 working. I could get
> the initial setup working, but after adding wifi configuration and doing a
> restart, it was lost. I finally gave up and installed 2.1.7

I downloaded the file from and uploaded the file manually in the UI.
Without having any issues. The software is running fine

rtollevik, 2022-10-17:
It happened also after a full reset.  My wifi is in access mode,  don't know if that have anything to do with it.  I could try it in router mode if you could make the v2.1.9 binary accessible somewhere.  I can reach the reader when I configure my mobile hotspot with the exact same ssid and password. For the record, my wifi is tp-link deco mesh, with telia as internet provider.

gskjold, 2022-10-17:
Attached is a ESP32-S2 firware where I have made a slight modification that may make a difference. If any of you could try it that would be great.

[esp32s2.zip](https://github.com/gskjold/AmsToMqttBridge/files/9798640/esp32s2.zip)

rtollevik, 2022-10-17:
I can try, but need some guidance.  Can I upgrade using this file, or should it be one of the .bin within?

gskjold, 2022-10-17:
Extract and use the firmware.bin file inside

ArnieO, 2022-10-17:
> It happened also after a full reset. My wifi is in access mode, don't know if that have anything to do with it. I could try it in router mode if you could make the v2.1.9 binary accessible [...]

To understand the issue, it would for sure be interesting if you could do a test with your mesh network in Router mode with same firmware. @gskjold : can you provide a link to 2.1.9 for such a test?

I also have a tp-link deco mesh system, in router mode. I have updated two devices to 2.1.9 ; no issues.

gskjold, 2022-10-17:
The original 2.1.9 firmware:
[ams2mqtt-esp32-2.1.9.zip](https://github.com/gskjold/AmsToMqttBridge/files/9798936/ams2mqtt-esp32-2.1.9.zip)
[ams2mqtt-esp8266-2.1.9.zip](https://github.com/gskjold/AmsToMqttBridge/files/9798937/ams2mqtt-esp8266-2.1.9.zip)
[ams2mqtt-esp32s2-2.1.9.zip](https://github.com/gskjold/AmsToMqttBridge/files/9798938/ams2mqtt-esp32s2-2.1.9.zip)

larserhardsen, 2022-10-17:
As soon as I saw the release email last night, I initiated the upgrade through the UI, and the unit did not seem to come back online.
However, after considering for some time to go outside and yank the Pow-K+ out of the Kamstrup meter (we had quite strong winds here, so the thought of going outside was not something I really felt like doing), somehow it resurfaced and has been stable since.

SVH-Powel, 2022-10-17:
> I upgraded using the GUI link.

Yes, I did also start with the gui link. I think I waited 5 minutes for it to come online. Then I flashed it with esptool.py without success. I then used esptool to erase the flash and I flashed 2.1.9 again. No success. I was able to enter the wifi configuration after the erase, but it never connected to my wifi after the reboot. I then flashed my previous version, 2.1.7 and it was up and running as soon as wifi configuration was entered and a reboot done.

gskjold, 2022-10-17:
> somehow it resurfaced and has been stable since.

Interesting, that could mean that this problem could appear suddenly for users that have already installed 2.1.9... I think I will make a new version 2.1.10 where I revert the changes around WiFi, then I have more time to figure out the problem.

gskjold, 2022-10-17:
@SVH-Powel Do you also have Pow-K+?

gskjold, 2022-10-17:
Here is a version that has all the changes for 2.1.9, but WiFi_connect() have been reverted:

[esp32s2.zip](https://github.com/gskjold/AmsToMqttBridge/files/9799843/esp32s2.zip)

Just to be certain that this actually is the problem, does this firmware fix your issue?

SVH-Powel, 2022-10-17:
> @SVH-Powel Do you also have Pow-K+?

No, it is a generic esp32

gskjold, 2022-10-17:
> No, it is a generic esp32

OK thanks, was just curious if this was isolated to ESP32-S2.
Here is the same version I posted above, just for ESP32:
[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/9799873/esp32.zip)

rtollevik, 2022-10-17:
Apparently ok with this release, identifying itself as 8ac1e03

stigvi, 2022-10-17:
> > No, it is a generic esp32
> 
> OK thanks, was just curious if this was isolated to ESP32-S2. Here is the same version I posted above, just for ESP32: [esp32.zip]

I tried it and it works as expected.

gskjold, 2022-10-17:
What hardware specifically @ArnieO  ?

retrov, 2022-10-17:
Running [AMS reader v2.1.10] now and seems fine on my network.

oz1iio, 2022-10-17:
Its still the same error. This morning’s I updated to v2.1.9 and it was stop reading the meter and were slow 
Now I updated to v2.1.10 and it’s still the same issue, it’s reset again and agin 
[image]
I use a kamstrup and a esp32 board. 
Anker

tronde-ams, 2022-10-17:
Both 2.1.9 and 2.1.10 have been OK for me. Aidon 3p IT, Chinese ESP32 configured as Lolin D32 and _no_ mesh WiFi. MQTT active.

oz1iio, 2022-10-18:
This morning is the same, up few seconds. 
[image]
V2.1.8 and back has worked grate, and I use it together with  home assisted

gskjold, 2022-10-18:
@oz1iio Do you also use HomeAssistant payload? Trying to find a pattern

gskjold, 2022-10-18:
This sounds like a different error than the original problem btw. So I am curious to know if this only happens when it is connected to the meter, or if it also happens when unplugged.

gskjold, 2022-10-18:
You know what, for those who still have problems with 2.1.10, fill out this template:
Country:
Meter:
Hardware:
MQTT payload:

ArnieO, 2022-10-18:
No problem here with 2.1.10 (up 16 hours)
- Norway
- Kamstrup
- Pow-K
- MQTT payload: JSON

retrov, 2022-10-18:
Seeing it being restarted after 3-5 hours after 2.1.10 install last night.

    Norway
    Kamstrup
    Pow-K
    MQTT payload: Home-Assistant

knakkwurst, 2022-10-18:
Instability (system hangs when entering mqtt config) on 2.1.9 and 2.1.10
In HA, I am missing energy column every 4th hour.
Norway
Aidon 6525
Pow-U
MQTT : Home Assistant

retrov, 2022-10-18:
I've reverted back to [AMS reader b420a0e] that was stable and had all the sensors correctly in Home Assistant.

stigvi, 2022-10-18:
I saw that config hours was set to 68 after I upgraded to the latest version. Could it be that who is slowing it down on your unit?

ArnieO, 2022-10-18:
What is "config hours"?

revellion, 2022-10-19:
I have issues with connecting to the PowK+ since upgrading from 2.1.8.

Though it could also be to poor wifi conditions with the meter on a pole behind the garage.

Country: Sweden 
Meter: Kamstrup 
Hardware: Pow-K+ (ESP32S2)
MQTT payload: Home Assistant

ArnieO, 2022-10-19:
> I saw that "average of top" X hours was set to 68 after I upgraded to the latest version. Could it be that which is slowing it down on your unit?

I found now that mine was also very high after the upgrade:
[image]
I changed it back to 3, and that value sticks.

retrov, 2022-10-19:
> I've reverted back to [AMS reader [b420a0e](https://github.com/gskjold/AmsToMqttBridge/commit/b420a0e6f4d3dabeb6071c2e3fbc2d595a0cd4b9)] that was stable and had all the sensors correctly in Home Assistant.

My uptime is a day now, so this one seems stable compared to the later builds I tried.

revellion, 2022-10-19:
> > I saw that "average of top" X hours was set to 68 after I upgraded to the latest version. Could it be that which is slowing it down on your unit?
> 
> I found now that mine was also very high after the upgrade: [image] I changed it back to 3, and that value sticks.

Did the same on my PowK+ with 2.1.10 on. And it made it stop rebooting. Which i suspected since I got WiFi signal measurements in MQTT but no meter payloads before it rebooted again. The web UI was never reachable. But if I removed the module from the meter then it was accessible

gskjold, 2022-10-19:
This is very interesting!

I really appreciate you all for testing and giving good feedback like this. I now have a better idea on what to test and hopefully reproduce this issue at home.

Stewart555, 2022-10-19:
Hi there,
Just want to add my experience with v 2.1.10 so far. It seems to be related to this 2.1.9 bug (?) and I rather add it to this thread than add a new one.

While using "ams-leser.local" to access the web-gui, the connection times out. Refreshing the page 4 or 5 times brings the page up. When then for example accessing the "Meter"-menu, it times out again. Sometimes the ESP icon is red and the other ones are grayed out.
If I access the reader via its IP-address, there are no connection issues.
(Maybe this is an issue with my network setup, but accessing the reader through the mDNS address was never a problem before)
MQTT-payload to HA, seems to be working fine.

The uptime is 3hrs now, but I did not reboot the reader since I updated it 2 days ago.

Country: Norway
Meter: Kamstrup
Hardware: Pow-K+ (ESP32S2)
MQTT payload: Home Assistant

revellion, 2022-10-20:
Started losing connection again. I'm gonna try resetting my PowK+ to factory and reconfigure

gskjold, 2022-10-20:
I have not been able to reproduce the problem locally, but I have made a variety of code improvements in the following firmware:
[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/9832848/esp32.zip)
[esp32s2.zip](https://github.com/gskjold/AmsToMqttBridge/files/9832849/esp32s2.zip)
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/9832850/esp8266.zip)

WARNING: This is to be considered pre-release firmware. If any firmware fault occurs you will have to be able to erase and re-flash via USB.

revellion, 2022-10-20:
Removed my PowK from the meter and testing the firmware above and see how it fares

gskjold, 2022-10-23:
Any feedback on the latest build

retrov, 2022-10-23:
Installed it now, will let you know the uptime later today. Previous versions seemed to only last 3-4 hours.

revellion, 2022-10-23:
I still have issues with getting data from it..but I'm gonna try and Supply a better wifi uplink for my powK. To rule out issues with wifi reception being the culprit. Though it's strange it worked a bit better on 2.1.8 in the past. Though it could be due to weather being more moist and cold causing issues. And the fact my meter is in a metal box on a pole. Not the best RF environment.

Maybe a new more RF friendly cover is needed and or some external antenna arrangement

ArnieO, 2022-10-23:
Still no issue with 2.1.10, so I stick with it for the time being.
[image]

ArnieO, 2022-10-23:
> I still have issues with getting data from it..but I'm gonna try and Supply a better wifi uplink for my powK. To rule out issues with wifi reception being the culprit. Though it's strange it worked [...]

Yes, good approach to try to isolate issues. I have -78 dBm RSSI, and no issues at that (relatively low) level; Pow-K+ inside closed metal fuse cabinet. Your situation sounds more challenging.

tronde-ams, 2022-10-23:
d8e265b is OK with me. Aidon 3p IT and MQTT raw. ESP32 configured as Lolin d32. No mesh wifi.

revellion, 2022-10-23:
> Yes, good approach to try to isolate issues. I have -78 dBm RSSI, and no issues at that (relatively low) level; Pow-K+ inside closed metal fuse cabinet. Your situation sounds more challenging.

Yeah likely, it's around -85/-90 RSSI sometimes so not really ideal. dunno if the firmware is set to watchdog reboot on that low RSSI likely.

revellion, 2022-10-23:
Upgraded to 2.1.11 and did a factory reset. And only reconfigured WiFi and meter settings and skipped the MQTT config..so far it's reachable

revellion, 2022-10-23:
Restored MQTT so far 42min uptime. Haven't restored EOTSE settings yet.

retrov, 2022-10-23:
> Any feedback on the latest build?

Running [d8e265b] with 9 hours uptime now. Seems more stable than when I was running 2.1.10

revellion, 2022-10-24:
Up for 11 hours now.

retrov, 2022-10-24:
> > Any feedback on the latest build?
> 
> Running [[d8e265b](https://github.com/gskjold/AmsToMqttBridge/commit/d8e265b7accf729b39c6cf56e11b28ac54cc5ae4)] with 9 hours uptime now. Seems more stable than when I was running 2.1.10

Checked my uptime just now and I see it's only 3 hours, so there has been a restart since my post yesterday.

tronde-ams, 2022-10-24:
About prices:

This is log from my 2.1.11 after a restart:

(18:42:12.716) <ESC>[0m(I) (EntsoeApi) Fetching prices for today
(18:42:12.756) <ESC>[0m(D) (EntsoeApi)  url: https://transparency.entsoe.eu/api?securityToken="MY TOKEN"&documentType=A44&periodStart=202210232200&p<ESC>[0meriodEnd=202210242200&in_Domain=10YNO-1--------2&out_Domain=10YNO-1--------2

This puzzles me:
Start=202210**23**2200
End=202210**24**2200

Today is 20221024, so I would expect End to read End=202210**25**2200

My 2.1.7 (without logging) does not fetch prices for tomorrow, neither does a friends with FW from earlier this summer (does not  exact version). all are Chinese ESP32 configured as Lolin D32.

ArnieO, 2022-10-24:
Apparently data is missing from Entso-E today, so let us focus elsewhere. 😃

tronde-ams, 2022-10-24:
I have a feeling at least some it is related to the code. See my previous post about date.

Last request before midnight

```
(23:56:34.588) <ESC>[0m(I) (EntsoeApi) Fetching prices for tomorrow
(23:56:36.627) <ESC>[0m(D) (EntsoeApi)  url: https://transparency.entsoe.eu/api?securityToken="MY TOKEN"&documentType=A44&periodStart=202210242200&p<ESC>[0meriodEnd=202210252200&in_Domain=10YNO-1--------2&out_Domain=10YNO-1--------2
(23:56:36.627) <ESC>[0m(D) (EntsoeApi)Connection established
(23:56:36.627) <ESC>[0m(D) (EntsoeApi)Receiving data
(23:56:38.029) <ESC>[0m(V) (unwrapData)(C1) HDLC frame:
(23:56:38.033) <ESC>[0m(V) 7E A0 2A 41  08 83 13 04  13 E6 E7 00  0F 40 00 00 <CR>
(23:56:38.048) <ESC>[0m(V) 00 00 01 01  02 03 09 06  01 00 01 07  00 FF 06 00 <CR>
(23:56:38.048) <ESC>[0m(V) 00 06 7E 02  02 0F 00 16  1B 72 AD 7E  <CR>
(23:56:38.062) <ESC>[0m(V) (unwrapData)(C1) LLC frame:
(23:56:38.062) <ESC>[0m(V) E6 E7 00 0F  40 00 00 00  00 01 01 02  03 09 06 01 <CR>
(23:56:38.062) <ESC>[0m(V) 00 01 07 00  FF 06 00 00  06 7E 02 02  0F 00 16 1B <CR>
(23:56:38.062) <ESC>[0m(V) <CR>
(23:56:38.062) <ESC>[0m(V) (unwrapData)(C1) DLMS frame:
(23:56:38.066) <ESC>[0m(V) 0F 40 00 00  00 00 01 01  02 03 09 06  01 00 01 07 <CR>
(23:56:38.084) <ESC>[0m(V) 00 FF 06 00  00 06 7E 02  02 0F 00 16  1B <CR>
(23:56:38.084) <ESC>[0m(V) (readHanPort)(C1) Received valid DLMS at 18
(23:56:38.084) <ESC>[0m(V) (readHanPort)(C1) Using application data:
(23:56:38.084) <ESC>[0m(V) 01 01 02 03  09 06 01 00  01 07 00 FF  06 00 00 06 <CR>
(23:56:38.088) <ESC>[0m(V) 7E 02 02 0F  00 16 1B <CR>
```
Midnight
```
(23:59:54.275) <ESC>[0m(V) (readHanPort)(C1) Received valid DLMS at 18
(23:59:54.275) <ESC>[0m(V) (readHanPort)(C1) Using application data:
(23:59:54.275) <ESC>[0m(V) 01 01 02 03  09 06 01 00  01 07 00 FF  06 00 00 06 <CR>
(23:59:54.300) <ESC>[0m(V) 77 02 02 0F  00 16 1B <CR>
(23:59:54.300) <ESC>[0m(V) (EnergyAccounting) Adding 0.0011 kWh import
(23:59:54.549) <ESC>[0m(V) (EnergyAccounting)  and 0.0000 NOK
(23:59:54.549) <ESC>[0m(V) (EnergyAccounting)  calculating threshold, currently at 0
(23:59:54.582) <ESC>[0m(V) (EnergyAccounting)  new threshold 0
(23:59:54.582) <ESC>[0m(D) (loop)(C1) Used 0 ms to update temperature
(23:59:54.582) <ESC>[0m(V) (unwrapData)(C1) HDLC frame:
(23:59:56.721) <ESC>[0m(V) 7E A0 2A 41  08 83 13 04  13 E6 E7 00  0F 40 00 00 <CR>
(23:59:56.741) <ESC>[0m(V) 00 00 01 01  02 03 09 06  01 00 01 07  00 FF 06 00 <CR>
(23:59:56.741) <ESC>[0m(V) 00 06 75 02  02 0F 00 16  1B F0 DB 7E  <CR>
(23:59:56.741) <ESC>[0m(V) (unwrapData)(C1) LLC frame:
(23:59:56.741) <ESC>[0m(V) E6 E7 00 0F  40 00 00 00  00 01 01 02  03 09 06 01 <CR>
(23:59:56.771) <ESC>[0m(V) 00 01 07 00  FF 06 00 00  06 75 02 02  0F 00 16 1B <CR>
(23:59:56.771) <ESC>[0m(V) <CR>
(23:59:56.771) <ESC>[0m(V) (unwrapData)(C1) DLMS frame:
(23:59:56.771) <ESC>[0m(V) 0F 40 00 00  00 00 01 01  02 03 09 06  01 00 01 07 <CR>
(23:59:56.771) <ESC>[0m(V) 00 FF 06 00  00 06 75 02  02 0F 00 16  1B <CR>
(23:59:56.771) <ESC>[0m(V) (readHanPort)(C1) Received valid DLMS at 18
(23:59:56.771) <ESC>[0m(V) (readHanPort)(C1) Using application data:
(23:59:56.796) <ESC>[0m(V) 01 01 02 03  09 06 01 00  01 07 00 FF  06 00 00 06 <CR>
(23:59:56.796) <ESC>[0m(V) 75 02 02 0F  00 16 1B <CR>
(23:59:56.796) <ESC>[0m(V) (EnergyAccounting) Adding 0.0011 kWh import
(23:59:57.046) <ESC>[0m(V) (EnergyAccounting)  and 0.0000 NOK
(23:59:57.046) <ESC>[0m(V) (EnergyAccounting)  calculating threshold, currently at 0
(23:59:57.046) <ESC>[0m(V) (EnergyAccounting)  new threshold 0
(23:59:57.046) <ESC>[0m(I) (EntsoeApi) Rotating price objects at ld
(23:59:59.408) <ESC>[0m(I) (EntsoeApi) Setting midnight millis lu
(23:59:59.433) <ESC>[0m(V) (unwrapData)(C1) HDLC frame:
(00:00:00.250) <ESC>[0m(V) 7E A1 0B 41  08 83 13 FA  7C E6 E7 00  0F 40 00 00 <CR>
(00:00:00.305) <ESC>[0m(V) 00 00 01 0C  02 02 09 06  01 01 00 02  81 FF 0A 0B <CR>
(00:00:00.305) <ESC>[0m(V) 41 49 44 4F  4E 5F 56 30  30 30 31 02  02 09 06 00 <CR>
(00:00:00.305) <ESC>[0m(V) 00 60 01 00  FF 0A 10 37  33 35 39 39  39 32 38 39 <CR>
(00:00:00.305) <ESC>[0m(V) 33 36 32 33  35 31 35 02  02 09 06 00  00 60 01 07 <CR>
(00:00:00.305) <ESC>[0m(V) FF 0A 04 36  35 32 35 02  03 09 06 01  00 01 07 00 <CR>
(00:00:00.305) <ESC>[0m(V) FF 06 00 00  06 66 02 02  0F 00 16 1B  02 03 09 06 <CR>
(00:00:00.305) <ESC>[0m(V) 01 00 02 07  00 FF 06 00  00 00 00 02  02 0F 00 16 <CR>
(00:00:00.305) <ESC>[0m(V) 1B 02 03 09  06 01 00 03  07 00 FF 06  00 00 00 23 <CR>
(00:00:00.305) <ESC>[0m(V) 02 02 0F 00  16 1D 02 03  09 06 01 00  04 07 00 FF <CR>
(00:00:00.355) <ESC>[0m(V) 06 00 00 00  00 02 02 0F  00 16 1D 02  03 09 06 01 <CR>
(00:00:00.355) <ESC>[0m(V) 00 1F 07 00  FF 10 00 20  02 02 0F FF  16 21 02 03 <CR>
[... 268 lines cut ...]
(00:07:35.662) <ESC>[0m(V) (EnergyAccounting)  calculating threshold, currently at 0
(00:07:35.694) <ESC>[0m(V) (EnergyAccounting)  new threshold 0
(00:07:35.756) <ESC>[0m(V) (unwrapData)(C1) HDLC frame:
(00:07:36.726) <ESC>[0m(V) 7E A0 2A 41  08 83 13 04  13 E6 E7 00  0F 40 00 00 <CR>
(00:07:36.746) <ESC>[0m(V) 00 00 01 01  02 03 09 06  01 00 01 07  00 FF 06 00 <CR>
(00:07:36.746) <ESC>[0m(V) 00 09 4E 02  02 0F 00 16  1B 8A CA 7E  <CR>
(00:07:36.746) <ESC>[0m(V) (unwrapData)(C1) LLC frame:
(00:07:36.771) <ESC>[0m(V) E6 E7 00 0F  40 00 00 00  00 01 01 02  03 09 06 01 <CR>
(00:07:36.771) <ESC>[0m(V) 00 01 07 00  FF 06 00 00  09 4E 02 02  0F 00 16 1B <CR>
(00:07:36.771) <ESC>[0m(V) <CR>
(00:07:36.771) <ESC>[0m(V) (unwrapData)(C1) DLMS frame:
[... cut ...]
(00:07:36.771) <ESC>[0m(V) 00 FF 06 00  00 09 4E 02  02 0F 00 16  1B <CR>
(00:07:36.771) <ESC>[0m(V) (readHanPort)(C1) Received valid DLMS at 18
(00:07:36.796) <ESC>[0m(V) (readHanPort)(C1) Using application data:
(00:07:36.796) <ESC>[0m(V) 01 01 02 03  09 06 01 00  01 07 00 FF  06 00 00 09 <CR>
(00:07:36.796) <ESC>[0m(V) 4E 02 02 0F  00 16 1B <CR>

```

It fetched new prices.

Did restart my other 2.1.11 and id did also fetch new prices.

I have not re-started the 2.1.7. It does still not show prices (01:24). will wait and see until 14:00.

[3 more replies]

closed 2022-10-25 by gskjold