UtilitechAS/amsreader-firmware #331: Don know if it is a bug but i am getting negative production values

CHHE-ledtec, 2022-10-19:
**Describe the bug**
Don know if it is a bug but i am getting negative production values

**Expected behavior**
A clear description of what you expected to happen.

**Screenshots**
[image]

**Hardware information:**
 - Country: Sweden
 - Meter: Kamstrup omnipower CT
 - Encryption enabled no
 - AMS reader: Pow-k+
 

**Relevant firmware information:**
 - Version: 2.1.8]
 - MQTT: yes
 - HAN GPIO: 
 - 
[image]

 - HAN baud and parity: 
 -
[image]

gskjold, 2022-10-19:
Must be some sort of overflow or error. Can you explain a bit more about your production? Peak power, what is the expected production etc.

ArnieO, 2022-10-19:
Peak production power is there in config/meter: 255 kWp.

gskjold, 2022-10-19:
If that really is the case I suspect that the registers would overflow quite easily. But I will also have to check why negative values are actually possible, as it can't really happen. Maybe we can squeeze some more out of the registers by changing from int to uint

ArnieO, 2022-10-19:
**Off topic (so please do not comment):** 
This is still the most spectacular overflow error (number goes from high positive to high negative) I know of:
First launch of the Ariane 5 rocket had a bug in the flight control software causing overflow error. Flight computers were redundant (two independent systems), but software was same on both. 
https://youtu.be/wGeZEUjUKvc

CHHE-ledtec, 2022-10-19:
yhee production is not 255kWp

i have 2 meter  - 
meter1 is connected too 8x50kWp inverters =400kWp
meter2 is connected too 9x50kWp inverters = 450kWp

4080 units of solarcells with 255Wp per panel.

http://www.windon.se/solpark.htm

soo possible errorbecuse wrong production value

gskjold, 2022-10-19:
Cool, I like it :) But probably overflowing, I will have a look if there is something we can do about it. How much do you produce on a nice summers day?

ArnieO, 2022-10-19:
Just a thought: Is the power meter really dimensioned for this power level? Could it be overflow in the meter?
I believe meters on large "industrial" installations often use special current transformers to divide down the measurements. (AmsToMqttBridge firmware has multiplicators that can be set to take that into account).

gskjold, 2022-10-20:
I have changed from int16 to uint 16, should allow 655kwh per bar in the graph

[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/9832848/esp32.zip)
[esp32s2.zip](https://github.com/gskjold/AmsToMqttBridge/files/9832849/esp32s2.zip)
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/9832850/esp8266.zip)

WARNING: This is to be considered pre-release firmware. If any firmware fault occurs you will have to be able to erase and re-flash via USB.

gskjold, 2022-11-01:
Closed due to inactivity