UtilitechAS/amsreader-firmware #332: display production instead of usage

CHHE-ledtec, 2022-10-19:
display production instead of usage 

[image]

gskjold, 2022-10-19:
Production should be a separate bar in the graph in the negative direction, but it does not seem to work properly...

Could you go to System -> Configuration file, unselect everything and download. This will export a file with all the values for the graph and real time data that I can then experiment with.

CHHE-ledtec, 2022-10-19:
CONFIGFILE.CONF
> amsconfig
> version v2.1.8
> boardType 5
> dayplot 4 1666180824 47479899 5320 5350 5330 5320 5320 5000 0 0 0 0 0 0 0 0 0 0 8230 5270 5280 5320 5320 5300 5240 5300 4294967295 -47830 -47830 -47830 -47830 -47830 -47630 82470 124700 -11940 -253440 -222320 -231490 -84480 -49150 -112830 -47830 -47830 -47830 -47830 -47830 -47830 -47830 -47830 -47830
> monthplot 5 1666130425 47437700 0 0 0 0 0 0 0 0 0 0 0 0 81430 76800 80770 77070 80370 77530 0 0 0 0 0 0 0 0 0 0 0 0 0 4294967295 0 0 0 0 0 0 0 0 0 0 0 0 -17410 256420 -188160 230900 -66100 -47830 0 0 0 0 0 0 0 0 0 0 0 0 0
> energyaccounting 4 10 0.00 0.00 0.00 0.00 12 9.80 13 9.92 14 9.60 15 9.28 16 8.95
>