UtilitechAS/amsreader-firmware #333: Homeassistant mqtt uniqe id endret?

gvenaas, 2022-10-19:
Ved siste oppgradering til 2.1.10 mistet Homeassistant "Energy" telleren.

Ser at uniq_id er endret i mqtt...
Fra: sensor.ams_esp32_tpi
Til: sensor.ams_esp32_tPI

Altså fra alle små til noen store bokstaver...

Bug eller ??

gskjold, 2022-10-20:
Sjekket i koden nå og jeg kan ikke se at den skulle ha vært lower case i tidligere versjoner, men jeg skal ta å teste dette ved anledning.

gvenaas, 2022-10-20:
Får dette i HA loggen.

2022-10-20 14:00:47.202 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'value_json' is undefined when rendering '{{ value_json.tPI | is_defined }}'
2022-10-20 14:00:47.207 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'value_json' is undefined when rendering '{{ value_json.tPO | is_defined }}'
2022-10-20 14:00:47.209 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'value_json' is undefined when rendering '{{ value_json.tQI | is_defined }}'
2022-10-20 14:00:47.210 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'value_json' is undefined when rendering '{{ value_json.tQO | is_defined }}'

gskjold, 2022-10-20:
Vanskelig å tolke uten å faktisk se dataene den mottok, men jeg vet desverre ikke nok om HA til å si noe om hvordan vi kan få sett det. Men jeg skal sjekke rundt heltimen hos meg hva som kommer ut.

gskjold, 2022-10-20:
Se der ja, "rtc":ld, her er det noe tull... sikkert derfor det ikke virker.

gskjold, 2022-10-21:
Fix kommer på neste versjon, testfirmware her:
[esp32.zip](https://github.com/gskjold/AmsToMqttBridge/files/9841018/esp32.zip)
[esp32s2.zip](https://github.com/gskjold/AmsToMqttBridge/files/9841019/esp32s2.zip)
[esp8266.zip](https://github.com/gskjold/AmsToMqttBridge/files/9841020/esp8266.zip)

gvenaas, 2022-10-21:
Det gjorde susen. Takk for kjaspp respons.

closed 2022-10-22 by gskjold