UtilitechAS/amsreader-firmware #335: Device arping all the time

bardahlm, 2022-10-20:
**Lots of arp**

Using tcpdump I see that the AMS arps every 10 seconds or so:

22:41:00.328958 ARP, Request who-has 10.88.88.70 tell 10.88.88.118, length 46
22:41:00.329009 ARP, Reply 10.88.88.70 is-at 3c:97:0e:3d:d1:23, length 28
22:41:01.565037 ARP, Request who-has 10.88.88.118 tell 10.88.88.106, length 46
22:41:03.271528 ARP, Request who-has 10.88.88.65 tell 10.88.88.118, length 46
22:41:11.415125 ARP, Request who-has 10.88.88.118 tell 10.88.88.118, length 46
22:41:12.272680 ARP, Request who-has 10.88.88.65 tell 10.88.88.118, length 46
22:41:12.343315 ARP, Request who-has 10.88.88.70 tell 10.88.88.118, length 46
22:41:12.343362 ARP, Reply 10.88.88.70 is-at 3c:97:0e:3d:d1:23, length 28
22:41:15.233579 ARP, Request who-has 10.88.88.106 tell 10.88.88.118, length 46
22:41:15.272984 ARP, Request who-has 10.88.88.106 tell 10.88.88.118, length 46
22:41:23.415856 ARP, Request who-has 10.88.88.118 tell 10.88.88.118, length 46
22:41:24.272414 ARP, Request who-has 10.88.88.65 tell 10.88.88.118, length 46
22:41:27.221838 ARP, Request who-has 10.88.88.106 tell 10.88.88.118, length 46
22:41:44.926736 ARP, Request who-has 10.88.88.70 tell 10.88.88.118, length 46
22:41:44.926783 ARP, Reply 10.88.88.70 is-at 3c:97:0e:3d:d1:23, length 28
22:41:45.273386 ARP, Request who-has 10.88.88.106 tell 10.88.88.118, length 46
22:41:45.274061 ARP, Request who-has 10.88.88.65 tell 10.88.88.118, length 46
22:41:45.274799 ARP, Request who-has 10.88.88.106 tell 10.88.88.118, length 46
22:41:51.088508 ARP, Request who-has 10.88.88.118 tell 10.88.88.70, length 28
22:41:51.089712 ARP, Reply 10.88.88.118 is-at a8:48:fa:c1:e0:76, length 46
22:41:56.019244 ARP, Request who-has 10.88.88.118 tell 10.88.88.118, length 46
22:41:56.086267 ARP, Request who-has 10.88.88.70 tell 10.88.88.118, length 46
22:41:56.086309 ARP, Reply 10.88.88.70 is-at 3c:97:0e:3d:d1:23, length 28
22:41:57.223808 ARP, Request who-has 10.88.88.106 tell 10.88.88.118, length 46
22:41:57.274638 ARP, Request who-has 10.88.88.106 tell 10.88.88.118, length 46
22:41:57.275314 ARP, Request who-has 10.88.88.65 tell 10.88.88.118, length 46
22:42:07.224980 ARP, Request who-has 10.88.88.118 tell 10.88.88.118, length 46
22:42:08.099551 ARP, Request who-has 10.88.88.70 tell 10.88.88.118, length 46
22:42:08.099601 ARP, Reply 10.88.88.70 is-at 3c:97:0e:3d:d1:23, length 28
22:42:09.229423 ARP, Request who-has 10.88.88.106 tell 10.88.88.118, length 46
22:42:09.273651 ARP, Request who-has 10.88.88.65 tell 10.88.88.118, length 46
22:42:09.274143 ARP, Request who-has 10.88.88.65 tell 10.88.88.118, length 46
22:42:30.960894 ARP, Request who-has 10.88.88.65 tell 10.88.88.118, length 46
22:42:30.961579 ARP, Request who-has 10.88.88.70 tell 10.88.88.118, length 46
22:42:30.961619 ARP, Reply 10.88.88.70 is-at 3c:97:0e:3d:d1:23, length 28
22:42:41.624761 ARP, Request who-has 10.88.88.118 tell 10.88.88.118, length 46
22:42:42.130449 ARP, Request who-has 10.88.88.70 tell 10.88.88.118, length 46
22:42:42.130484 ARP, Reply 10.88.88.70 is-at 3c:97:0e:3d:d1:23, length 28
22:42:42.961510 ARP, Request who-has 10.88.88.65 tell 10.88.88.118, length 46
22:42:44.230244 ARP, Request who-has 10.88.88.106 tell 10.88.88.118, length 46

(10.88.88.118 is the AMS, .70 is the server with MQTT and the machine doing tcpdump, .70 is my PC)

**To Reproduce**

Run tcpdump or other packet capture on the network

**Expected behavior**
The device should cache arp longer, 5 minutes for example.

**Hardware information:**
 - Country: Norway
 - Meter: Kamsptrup
 - Encryption enabled: no
 - AMS reader: POW-K

**Relevant firmware information:**
 - Version: betabuild from #330 
 - MQTT: yes
 - HAN GPIO: GPIO12
 - HAN baud and parity: 2400 8N1
 - ENTSO-E API enabled: no

**Additional context**
I discovered this while troubleshooting missing measurements and MQTT errors.

gskjold, 2022-10-21:
If I remember correctly, there is no ARP cache in the ESP network stack. Besides, that is out of my control. You are mentioning MQTT errors, please elaborate. Check telnet log and MQTT broker log for clues

bardahlm, 2022-10-21:
I'll open a separate issue for MQTT issues.

closed 2022-10-21 by bardahlm