UtilitechAS/amsreader-firmware #336: MQTT fails at times, missing data

bardahlm, 2022-10-21:
**Describe the bug**
Some time lately I observed that I'm not getting data as frequently as I'm used to.

Logging in to the GUI I sometimes see MQTT marked red.

Telnet drops connection shortly after connection (I suspected duplicate IP or other issues but ruled that out using tcpdump but found the device sends lots of arps (#335). )

``` 
# telnet 10.88.88.118
Trying 10.88.88.118...
Connected to 10.88.88.118.
Escape character is '^]'.
*** Remote debug - over telnet - for ESP8266 (NodeMCU) - version 3.0.5
* Host name: meter IP:10.88.88.118 Mac address:A8:48:FA:C1:E0:76
* Free Heap RAM: 14320
* ESP SDK version: 2.2.2-dev(38a443e)
******************************************************
* Commands:
    ? or help -> display these help of commands
    q -> quit (close this connection)
    m -> display memory available
    v -> set debug level to verbose
    d -> set debug level to debug
    i -> set debug level to info
    w -> set debug level to warning
    e -> set debug level to errors
    s -> set debug silence on/off
    l -> show debug level
    t -> show time (millis)
    profiler:
      p      -> show time between actual and last message (in millis)
      p min  -> show only if time is this minimal
      P time -> set debug level to profiler
    c -> show colors
    filter:
          filter <string> -> show only debugs with this
          nofilter        -> disable the filter
    cpu80  -> ESP8266 CPU a 80MHz
    cpu160 -> ESP8266 CPU a 160MHz

* Please type the command and press enter to execute.(? or h for this help)
***
(W) (errorBlink) MQTT connection not available, double blink
(V) (unwrapData) HDLC frame:
(V) 7E A0 E2 2B  21 13 23 9A  E6 E7 00 0F  00 00 00 00
(V) 0C 07 E6 0A  15 05 07 21  28 FF 80 00  00 02 19 0A
(V) 0E 4B 61 6D  73 74 72 75  70 5F 56 30  30 30 31 09
(V) 06 01 01 00  00 05 FF 0A  10 35 37 30  36 35 36 37
(V) 32 37 32 37  39 32 33 33  37 09 06 01  01 60 01 01
(V) FF 0A 12 36  38 34 31 31  33 31 42 4E  32 34 33 31
(V) 30 31 30 34  30 09 06 01  01 01 07 00  FF 06 00 00
(V) 03 20 09 06  01 01 02 07  00 FF 06 00  00 00 00 09
(V) 06 01 01 03  07 00 FF 06  00 00 00 00  09 06 01 01
(V) 04 07 00 FF  06 00 00 01  BB 09 06 01  01 1F 07 00
(V) FF 06 00 00  00 55 09 06  01 01 33 07  00 FF 06 00
(V) 00 00 A0 09  06 01 01 47  07 00 FF 06  00 00 00 F2
(V) 09 06 01 01  20 07 00 FF  12 00 EB 09  06 01 01 34
(V) 07 00 FF 12  00 ED 09 06  01 01 48 07  00 FF 12 00
(V) EC 66 30 7E
(V) (unwrapData) LLC frame:
(V) E6 E7 00 0F  00 00 00 00  0C 07 E6 0A  15 05 07 21
(V) 28 FF 80 00  00 02 19 0A  0E 4B 61 6D  73 74 72 75
(V) 70 5F 56 30  30 30 31 09  06 01 01 00  00 05 FF 0A
(V) 10 35 37 30  36 35 36 37  32 37 32 37  39 32 33 33
(V) 37 09 06 01  01 60 01 01  FF 0A 12 36  38 34 31 31
(V) 33 31 42 4E  32 34 33 31  30 31 30 34  30 09 06 01
(V) 01 01 07 00  FF 06 00 00  03 20 09 06  01 01 02 07
(V) 00 FF 06 00  00 00 00 09  06 01 01 03  07 00 FF 06
(V) 00 00 00 00  09 06 01 01  04 07 00 FF  06 00 00 01
(V) BB 09 06 01  01 1F 07 00  FF 06 00 00  00 55 09 06
[... 79 lines cut ...]
Oct 21 08:46:20 x230 mosquitto: 1666334780: Socket error on client amsmeter, disconnecting.
Oct 21 08:46:48 x230 mosquitto: 1666334808: New connection from 10.88.88.118 on port 1883.
Oct 21 08:46:48 x230 mosquitto: 1666334808: New client connected from 10.88.88.118 as amsmeter (p2, c1, k10, u'meter').
```

**To Reproduce**
TBD

**Expected behavior**
Data submitted to MQTT every time metrics come in from the meter.

**Screenshots**

**Hardware information:**
 - Country: Norway
 - Meter: Kamsptrup
 - Encryption enabled: no
 - AMS reader: POW-K

**Relevant firmware information:**
 - Version: betabuild from #330 330
 - MQTT: yes
 - HAN GPIO: GPIO12
 - HAN baud and parity: 2400 8N1
 - ENTSO-E API enabled: no

ArnieO, 2022-10-21:
@gskjold This is a Pow-K with ESP8266 sold in Nov-21. 

@bardahlm : No such issues with v2.1.8 or previous versions?

gskjold, 2022-10-21:
> I can no longer upgrade using web

To clearify, are you then uploading a bin file in System -> Firmware ?

gskjold, 2022-10-21:
Ignore that, I have found the problem, will fix :)

gskjold, 2022-10-21:
Question, when telnet is dropped, does it also reboot? In other words, how is the uptime on your device?
What payload type do you use for MQTT?

bardahlm, 2022-10-21:
Device do not reboot when telnet drops. Payload is json. Device runs for a few days or so before booting.

gskjold, 2022-10-21:
It sort of sounds like it is loosing wifi connection, any indication in logs on your wireless?

bardahlm, 2022-10-21:
```
Fri Oct 21 18:32:04 2022 | Client Excluded: MACAddress:a8:48:fa:c1:e0:76 Base Radio MAC :f0:7f:06:d7:dc:a0 Slot: 0 User Name: unknown Ip Address: 10.88.88.118 Reason:Attempted to use IP Address assigned to another device. ReasonCode: 3
```

This one looks interesting!

</body>
</html>

bardahlm, 2022-10-21:
I have allowed static IPs on my wlan, but for some reason it's not working. I changed to DHCP now.

bardahlm, 2022-10-21:
Problem caused by wifi issues.

closed 2022-10-21 by bardahlm