UtilitechAS/amsreader-firmware #337: Line graph showing power usage over time

shadowano, 2022-10-21:
**Is your feature request related to a problem? Please describe.**
Not a problem, only a request for a new feature to make the product better

**Describe the solution you'd like**
I would like to propose a new feature which shows a graph over time (time can be selected e.g. 5min, 30min, 1h, 3h, 6h) that shows the power usage over time. For better user interface/visibility it should be a line graph. I would love to see this feature because it would help me understand how my house appliances uses power, and for how long. Some energy intensive units would be easy to see in such a graph. 

**Describe alternatives you've considered**
Tibber app, but I haven't considered it. 

**Additional context**
One example from the Tibber app shows what I mean (stolen screenshot)
[image]

ArnieO, 2022-10-21:
Interesting proposition, and maybe to some degree feasible in ESP32 based devices. The challenge is storage space for the data, which from some meters arrives as often as each 2.5 second.

shadowano, 2022-10-22:
2.5 seconds updates, gives 24 data plots for 1h, 48 plots for 2 hours. Will that be a problem on ESP32 devices regarding storage space? Starting with 2h, or even a bit more would help a lot. 

Another thing that could be considered is to let users disable some graphs to free up storage space for such a feature, or other new features.  Users decide what to show based on what is possible storage wise. 

To extend it further (probably a topic in its own) support external storage (Google, Dropbox, azure blob storage, network file share etc).

bardahlm, 2022-10-22:
You need data for a year or so, atleast, so you can compare and do analysis. Set up MQTT and Influxdb, see https://www.influxdata.com/influxdb-cloud-and-mqtt/ for example.

shadowano, 2022-10-22:
> You need data for a year or so, atleast, so you can compare and do analysis. Set up MQTT and Influxdb, see https://www.influxdata.com/influxdb-cloud-and-mqtt/ for example.

For the feature I'm asking for you can throw away data older than what is set as max, e.g 2h.

gskjold, 2022-10-23:
This may be possible in runtime (data kept in RAM, never stored on flash) because of the simple fact that flash memory has a limited amount of write cycles. This would mean that data was lost on reboot, but I don't see that as a problem for this feature. External storage is already indirectly possible with MQTT. Writing to external file systems or cloud solutions are out of the question because of memory limitations.

shadowano, 2022-10-23:
> This may be possible in runtime (data kept in RAM, never stored on flash) because of the simple fact that flash memory has a limited amount of write cycles. This would mean that data was lost on reboot, but I don't see that as a problem for this feature. 

Good to hear, and I agree as long as the app is stable and not causing unnecessary reboots it shouldn't be a problem at all :)

gskjold, 2023-01-05:
It would require 720 datapoints for 10sec resolution over 2hrs, and if we use uint16 (65535w max) this adds up to 2.8kb if we are going to store data for both import and export. This would be ok for ESP32, but for ESP8266 it would be taking up too much resources. I may consider it for ESP32

bardahlm, 2023-01-06:
> store data for both import and export

Can a signed variable be used, with positive for import and negative for export, as there is never both at the same time?

ehorvat1, 2023-05-13:
Hi, I did a version with a 1 hour plot which holds 180 data points. Flash memory for ESP8266 gets maxed out with this...see:  https://github.com/ehorvat1/AmsToMqttBridge-EHo

gskjold, 2023-05-14:
Thanks, great to see some community implementations! I started this a while back for v2.3, but progress stalled as other things got prioritized... Maybe I'll get around to it again soon.

gskjold, 2023-05-14:
I had a quick look at what I had, and it turns out it only needed some adjustments on the user interface. Done the changes and made a build. Its a 360 point 1hr dataset stored in a int8 with a uint8 as scaling factor, which reduces precision a bit, but I think we will be good. Note that you may have to enable realtime plot in configuration.

Keep in mind that this is part of v2.3 development and may be unstable. Please only install if you have the possibility to downgrade via USB/FTDI. Always back up your configuration before upgrading to development.

[esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11472050/esp32solo.zip)
[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11472052/esp32.zip)
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11472053/esp32c3.zip)
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11472054/esp32s2.zip)
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11472055/esp8266.zip)

closed 2023-12-25 by gskjold